noble/noble #797: Disconnect peripheral outside 'discover' loop

NeoSyder, 2018-05-22:
I would like to know if is it possible to disconnect a peripheral outside the 'discover' listener. 

I tried so far to save my peripheral in a variable to call later but the object is always null or undefined in contrary to my characteristics.

Tell me if you want some code.

don, 2018-05-26:
@NeoSyder A simple example would be helpful

NeoSyder, 2018-05-28:
```
var peripheralSave = null;

function launchScan() {
  noble.on('stateChange', function(state) {
    if (state === 'poweredOn') {
        setTimeout(function() {
            console.log("scan terminé");
            noble.stopScanning();
        }, 10000);
        console.log("scan lancé");
        noble.startScanning([gaspardServiceUuid], false);
    } else {
      noble.stopScanning();
      //response.status(500).send("Bluetooth non connecté");
    }
  });
}

noble.on('discover', function(peripheral) {

  console.log("Name find " + peripheral.advertisement.localName);
  if (!found) {
    if (peripheral.advertisement.localName === gaspardName) {
      console.log("\ngaspard trouvé");
      found = true;
      noble.stopScanning();
      cancelTimeout = true;
      console.log("scan terminé");
      peripheral.connect(function(err) {
        if (err) {
          peripheral.disconnect();
        }
        this.peripheralSave = peripheral;
        this.gaspardNamePrinter = peripheral.advertisement.localName;
        peripheral.discoverServices([gaspardBatteryService, realtimeServiceUuid], function(err, services) {
          services.forEach(function(service) {
            console.log('found service:', service.uuid);
            if(service.uuid === gaspardBatteryService){
              service.discoverCharacteristics([batteryCharacteristicUuid], function(error, characteristics){
                characteristics[0].read(function(err, data){
                  if(err){
                    console.log(err);
                  }else{
                    console.log(data);
                  }
                });
              });
            }else if(service.uuid === realtimeServiceUuid){
              service.discoverCharacteristics([], function(err, characteristics) {
                characteristics.forEach(function(characteristic) {
                  console.log('found characteristic:', characteristic.uuid);
                  if (realtimeCharacteristicUuid == characteristic.uuid) {
                    realtimeCharacteristic = characteristic;
                  } else if (realtimeCharacteristicValueUuid == characteristic.uuid) {
                    realtimeCharacteristicValue = characteristic;
                  }
                });
                if (realtimeCharacteristic &&
                  realtimeCharacteristicValue) {
                  response.send({
                    message: 'Gaspard trouvé',
                    gaspardName: peripheral.advertisement.localName,
                  });
                } else {
                  peripheral.disconnect();
                  console.log('missing characteristics');
                  response.status(500).send("Charactéristiques non présentes...déconnexion");
                }
              });
            }
          });
        });
      });
    }
  }
});

function disconnect() {
  console.log('passe ici ' + peripheralSave);
  peripheralSave.disconnect(function(err) {
    if(err){
        console.log(err);
    }
  });
}
```

It's when i call disconnect in an other method, peripheral is null in my log but when I when I test with a characteristic it's not null or undefined

don, 2018-05-30:
@NeoSyder peripheralSave is never being set in your example. Try `peripheralSave` rather than `this. peripheralSave`. Update the UUID and try at the sample below.

	var noble = require('noble');
	var peripheralSave = null;
	var gaspardServiceUuid = '207E6EB0-FA9E-4ADC-8BC0-C113CD6755E7';

	noble.on('stateChange', function(state) {
	  if (state === 'poweredOn') {
	    setTimeout(function() {
	      console.log("scan terminé");
	      noble.stopScanning();
	    }, 10000);
	    console.log("scan lancé");
	    noble.startScanning([gaspardServiceUuid], false);
	  } else {
	    noble.stopScanning();
	  }
	});

	noble.on('discover', function(peripheral) {
	  noble.stopScanning();
	  peripheral.connect(function(err) {
	    console.log('connected to', peripheral.id);
	    peripheralSave = peripheral;
	    console.log('calling disconnect in 1 second');
	    setTimeout(disconnect, 1000);
	  });
	});

	function disconnect() {
	  console.log('disconnecting from', peripheralSave.id)
	  console.log('passe ici ' + peripheralSave);
	  peripheralSave.disconnect(function(err) {
	    if(err){
	      console.log(err);
	    }
	  });
	}

NeoSyder, 2018-06-05:
Thank you for your response. I tried that and it's work !

closed 2018-06-05 by don