noble/noble #798: Gatt becomes undefined?

agonz519, 2018-05-24:
Currently trying to get the Parrot Mambo mini-drone to take off and have been unsuccessful. 

[image]

The code that is showing this error is here in Binding.js:

` NobleBindings.prototype.write = function(peripheralUuid, serviceUuid, characteristicUuid, data, withoutResponse) {
var handle = this._handles[peripheralUuid];
var gatt = this._gatts[handle];

if (gatt) {
gatt.write(serviceUuid, characteristicUuid, data, withoutResponse);
} else {
console.warn('noble warning: unknown peripheral ' + peripheralUuid);
}
}; `

I have tried console.logging **gatt** and it shows a lot of information, but at some point while I'm waiting for the device to take off, gatt becomes undefined, and then the unknown peripheral error starts to show over and over. 

What exactly is gatt used for and any idea why it becomes undefined at some point? 

I know  my question is a little broad so please let me know what else I can provide.

agonz519, 2018-05-25:
I mean. It says connected. The lights on the mambo stop blinking. Then it seems to be waiting for a command. Then I start getting the unknown peripheral error and the lights start blinking again. So it looks like it disconnects at some point. What might cause this?

don, 2018-05-26:
@agonz519 I expect that the drone disconnects after you send the takeoff command. Are you handling disconnects? gatt represents the bluetooth service on the other side of the connection. https://www.bluetooth.com/specifications/gatt/generic-attributes-overview

agonz519, 2018-05-26:
@don Hey Don, thanks for the response. 
I'm not handling disconnects, I'm simply trying to run the "Get started in 60 seconds" example found here: 
https://www.npmjs.com/package/parrot-minidrone

I'll see about doing that though, maybe I can get it to reconnect before it starts spazzing out with unknown peripheral. Thanks for the documentation as well.