noble/noble #799: Cannot find module '../build/Release/binding.node'

Lvf974, 2018-05-25:
Hi, I tried to make this command to track communications of a connected object. And I have this mistake that I can't solve.

sudo node advertise.js -a devices/b4994c5f9e2d_AL-Bc7.adv.json -s devices/b4994c5f9e2d.srv.json 
module.js:549
    throw err;
    ^

Error: Cannot find module '../build/Release/binding.node'
    at Function.Module._resolveFilename (module.js:547:15)
    at Function.Module._load (module.js:474:25)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/home/ludovic/node_modules/gattacker/node_modules/bluetooth-hci-socket/lib/native.js:3:15)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)

don, 2018-05-26:
It sounds like something isn't built correctly. I'd remove `node_modules` in your project and run `npm install` again. If you still get error, please provide more info about the node version, os, architecture etc.

ujj-u, 2018-06-05:
I am also facing the same issue. Tried reinstalling everything.

I am on arch I installed every bluez package. Arch is absent from nobel's readme file. Adding dependencies for Arch would be helpful.

don, 2018-06-05:
@ujj-u we need more details. My guess is [bluetooth-hci-socket](https://github.com/noble/node-bluetooth-hci-socket) failed to compile. Make sure your system has all the dependencies. In addition to bluez you usually need libudev-dev, python2.7, make, gcc. I don't know anything about Arch but if you make suggestions, we can get the docs updated.

See if you can build bluetooth-hci-socket. I recommend using Node 8.x.

    cd ~
    git clone https://github.com/noble/node-bluetooth-hci-socket
    cd node-bluetooth-hci-socket
    npm install

If that works, try and build noble

    cd ~
    git clone https://github.com/noble/noble
    cd noble
    npm install
    cd examples/
    sudo node advertisement-discovery.js

Lvf974, 2018-06-06:
I managed to install, my node version was not the same as the nodejs version

closed 2018-06-08 by don

yesimxev, 2019-06-06:
For anyone who needs a solution. Looks like the bindings.node can be just simply pasted where the program looks for it. Eg. use `strace YOUR_COMMAND` and check which folder is it being searching in.
Then 
```
git clone https://github.com/noble/node-bluetooth-hci-socket.git
mkdir -p /root/YOUR_PROGRAM/node_modules/noble/node_modules/bluetooth-hci-socket/build/Release
cp /root/node-bluetooth-hci-socket/build/Release/binding.node /root/YOUR_PROGRAM/node_modules/noble/node_modules/bluetooth-hci-socket/build/Release/
```

sumitjaju, 2019-09-08:
> @ujj-u we need more details. My guess is [bluetooth-hci-socket](https://github.com/noble/node-bluetooth-hci-socket) failed to compile. Make sure your system has all the dependencies. In addition to [...]

Don, when I issed npm install for node-bluetooth-hci-socket project it gives error of 'platform unsupported'. Is macos not supported. 

`npm ERR! notsup Unsupported platform for bluetooth-hci-socket@0.5.2: wanted {"os":"linux,android,freebsd,win32","arch":"any"} (current: {"os":"darwin","arch":"x64"})`

Between I am trying to setup noble but gives me below error. Please advise. 

```
module.js:549
    throw err;
    ^

Error: Cannot find module './build/Release/binding.node'
    at Function.Module._resolveFilename (module.js:547:15)
    at Function.Module._load (module.js:474:25)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at Object.<anonymous> (/Users/Meet/Front-workspace/startupnode/node_modules/xpc-connection/index.js:3:15)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
```

bugs181, 2020-02-07:
Having the same issue (with almost the same specs) as @sumitjaju 

`node -v`: `v12.12.0`
`OS`: `OS X Mojave 10.14`

yesimxev, 2020-02-07:
Replacing the line in the .cpp worked for me, or you can replace the whole folder
https://github.com/abandonware/node-bluetooth-hci-socket/commit/9b9e9753a6e676754a851369f723c1bbcf7783c6

pvonmoradi, 2020-12-03:
Same issue for `v12.20.0` trying to get https://github.com/tralamazza/nus_console running.

solarkraft, 2022-05-23:
I have a similar looking error on an M1 Mac with node 16.15.0. 

Installing fails with: 

```
paul@KekBook noble-messing % yarn add noble
yarn add v1.22.18
warning package.json: No license field
warning package-lock.json found. Your project contains lock files generated by tools other than Yarn. It is advised not to mix package managers in order to avoid resolution inconsistencies caused by unsynchronized lock files. To clear this warning, remove package-lock.json.
warning No license field
[1/4] 🔍  Resolving packages...
[2/4] 🚚  Fetching packages...
[3/4] 🔗  Linking dependencies...
[4/4] 🔨  Building fresh packages...
warning Error running install script for optional dependency: "/Users/paul/Documents/JawboneHacking/noble-messing/node_modules/xpc-connection: Command failed.
Exit code: 1
Command: node-gyp rebuild
Arguments:
Directory: /Users/paul/Documents/JawboneHacking/noble-messing/node_modules/xpc-connection
Output:
gyp info it worked if it ends with ok
gyp info using node-gyp@9.0.0
gyp info using node@16.15.0 | darwin | arm64
gyp info find Python using Python version 3.8.9 found at \"/Library/Developer/CommandLineTools/usr/bin/python3\"
gyp info spawn /Library/Developer/CommandLineTools/usr/bin/python3
gyp info spawn args [
gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/paul/Documents/JawboneHacking/noble-messing/node_modules/xpc-connection/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/Users/paul/Library/Caches/node-gyp/16.15.0/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/Users/paul/Library/Caches/node-gyp/16.15.0',
gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/Users/paul/Library/Caches/node-gyp/16.15.0/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/Users/paul/Documents/JawboneHacking/noble-messing/node_modules/xpc-connection',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
  CXX(target) Release/obj.target/binding/src/XpcConnection.o
../src/XpcConnection.cpp:26:77: error: too few arguments to function call, single argument 'context' was not specified
  target->Set(Nan::New(\"XpcConnection\").ToLocalChecked(), tmpl->GetFunction());
                                                          ~~~~~~~~~~~~~~~~~ ^
/Users/paul/Library/Caches/node-gyp/16.15.0/include/node/v8.h:6802:46: note: 'GetFunction' declared here
  V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
                                             ^
../src/XpcConnection.cpp:26:11: error: no matching member function for call to 'Set'
  target->Set(Nan::New(\"XpcConnection\").ToLocalChecked(), tmpl->GetFunction());
  ~~~~~~~~^~~
/Users/paul/Library/Caches/node-gyp/16.15.0/include/node/v8.h:3961:37: note: candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                    ^
/Users/paul/Library/Caches/node-gyp/16.15.0/include/node/v8.h:3964:37: note: candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                    ^
../src/XpcConnection.cpp:103:54: error: too few arguments to function call, single argument 'context' was not specified
    xpcObject = xpc_int64_create(value->IntegerValue());
                                 ~~~~~~~~~~~~~~~~~~~ ^
[... 179 lines cut ...]
- /Users/paul/Documents/JawboneHacking/noble-messing/node_modules/noble/lib/resolve-bindings.js
- /Users/paul/Documents/JawboneHacking/noble-messing/node_modules/noble/index.js
- /Users/paul/Documents/JawboneHacking/noble-messing/main.ts
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:933:15)
    at Function.Module._load (node:internal/modules/cjs/loader:778:27)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/Users/paul/Documents/JawboneHacking/noble-messing/node_modules/noble/lib/mac/highsierra.js:7:21)
    at Module._compile (node:internal/modules/cjs/loader:1105:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1159:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/Users/paul/Documents/JawboneHacking/noble-messing/node_modules/noble/lib/mac/highsierra.js',
    '/Users/paul/Documents/JawboneHacking/noble-messing/node_modules/noble/lib/mac/bindings.js',
    '/Users/paul/Documents/JawboneHacking/noble-messing/node_modules/noble/lib/resolve-bindings.js',
    '/Users/paul/Documents/JawboneHacking/noble-messing/node_modules/noble/index.js',
    '/Users/paul/Documents/JawboneHacking/noble-messing/main.ts'
  ]
}
```

Interestingly installation using npm only shows the node-gyp error when installing xpc-connection explicitly. The runtime error is the same.

maelp, 2024-10-10:
I still have this, on osX (the above module doesn't work for osX)