noble/noble #805: `noble` installation fails for `node` version `10` (Ubuntu 18.04)

spanierm, 2018-06-10:
I tried to use `noble` with `node` version `10.4.0` but the package installation failed on Ubuntu 18.04.

`noble` installation works as expected with `node` version `9.3.0` on Ubuntu 18.04. MacOS seems to have a simliar issue, cf. #791 .

### Recreate the issue
1. `node` version `10.4.0` installed
1. ``sudo setcap cap_net_raw+eip $(eval readlink -f `which node`)``
1. `npm install noble`

### Outcome
```txt
/tmp/noble-node10-error$ node --version
v10.4.0
/tmp/noble-node10-error$ npm install noble

> usb@1.3.1 install /tmp/noble-node10-error/node_modules/usb
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.3.1/usb_bindings-v1.3.1-node-v64-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for usb@1.3.1 and node@10.4.0 (node-v64 ABI, glibc) (falling back to source compile with node-gyp) 
make: Entering directory '/tmp/noble-node10-error/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
node-pre-gyp ERR! Tried to download(undefined): https://github.com/tessel/node-usb/releases/download/1.3.1/usb_bindings-v1.3.1-node-v64-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for usb@1.3.1 and node@10.4.0 (node-v64 ABI, glibc) (falling back to source compile with node-gyp) 
../libusb/libusb/core.c:2342:1: fatal error: opening dependency file ./Release/.deps/Release/obj.target/libusb/libusb/libusb/core.o.d.raw: No such file or directory
 }
 ^
compilation terminated.
libusb.target.mk:134: recipe for target 'Release/obj.target/libusb/libusb/libusb/core.o' failed
make: *** [Release/obj.target/libusb/libusb/libusb/core.o] Error 1
make: Leaving directory '/tmp/noble-node10-error/node_modules/usb/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/user/.nvm/versions/node/v10.4.0/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:237:12)
gyp ERR! System Linux 4.15.0-22-generic
gyp ERR! command "/home/user/.nvm/versions/node/v10.4.0/bin/node" "/home/user/.nvm/versions/node/v10.4.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/tmp/noble-node10-error/node_modules/usb/src/binding/usb_bindings.node" "--module_name=usb_bindings" "--module_path=/tmp/noble-node10-error/node_modules/usb/src/binding"
gyp ERR! cwd /tmp/noble-node10-error/node_modules/usb
gyp ERR! node -v v10.4.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/home/user/.nvm/versions/node/v10.4.0/bin/node /home/user/.nvm/versions/node/v10.4.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/tmp/noble-node10-error/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/tmp/noble-node10-error/node_modules/usb/src/binding' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/tmp/noble-node10-error/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:961:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:5)
node-pre-gyp ERR! System Linux 4.15.0-22-generic
node-pre-gyp ERR! command "/home/user/.nvm/versions/node/v10.4.0/bin/node" "/tmp/noble-node10-error/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /tmp/noble-node10-error/node_modules/usb
node-pre-gyp ERR! node -v v10.4.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.39
node-pre-gyp ERR! not ok 
Failed to execute '/home/user/.nvm/versions/node/v10.4.0/bin/node /home/user/.nvm/versions/node/v10.4.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/tmp/noble-node10-error/node_modules/usb/src/binding/usb_bindings.node --module_name=usb_bindings --module_path=/tmp/noble-node10-error/node_modules/usb/src/binding' (1)

> bluetooth-hci-socket@0.5.1 install /tmp/noble-node10-error/node_modules/bluetooth-hci-socket
> node-gyp rebuild

make: Entering directory '/tmp/noble-node10-error/node_modules/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::poll()’:
../src/BluetoothHciSocket.cpp:251:95: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated [-Wdeprecated-declarations]
     Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
                                                                                               ^
In file included from ../src/BluetoothHciSocket.cpp:8:0:
../../nan/nan.h:950:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::emitErrnoError()’:
[... 34 lines cut ...]
gyp ERR! not ok 
npm WARN saveError ENOENT: no such file or directory, open '/tmp/noble-node10-error/package.json'
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN enoent ENOENT: no such file or directory, open '/tmp/noble-node10-error/package.json'
npm WARN noble-node10-error No description
npm WARN noble-node10-error No repository field.
npm WARN noble-node10-error No README data
npm WARN noble-node10-error No license field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.3.1 (node_modules/usb):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.3.1 install: `node-pre-gyp install --fallback-to-build`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

+ noble@1.9.1
added 4 packages from 4 contributors and audited 278 packages in 3.778s
found 11 vulnerabilities (3 low, 8 moderate)
  run `npm audit fix` to fix them, or `npm audit` for details
```

### Expected outcome:
* Successful installation without the errors for `node-gyp`

jrobeson, 2018-06-17:
I can replicate this outside of a docker container as well.

jrobeson, 2018-06-18:
https://github.com/noble/node-bluetooth-hci-socket/issues/84

spanierm, 2018-06-19:
Thanks for the investigation, @jrobeson. I also suspected one of the upstream projects but was not able to investigate in all detail.

nathankellenicki, 2018-08-07:
Paging @sandeepmistry. Is this library still maintained?

robey, 2018-09-12:
the @sandeepmistry xpc-connection library appears to be abandonware. noble should add an explicit dependency on the PR that fixes it: https://github.com/sandeepmistry/node-xpc-connection/pull/26

LeviSchuck, 2018-09-14:
Abandonware huh. That's unfortunate.

jrobeson, 2018-09-15:
you folks could try my refactor PR if you feel like it. It does way more than fix the node 10 issue https://github.com/noble/noble/pull/823.  Alternatively you could force the bluetooth-hci-socket PR here: https://github.com/noble/node-bluetooth-hci-socket/pull/91

I've basically gone beyond all this and converted all of noble to typescript for my own personal usage.  I'm attempting to keep the public API the same , at least initially.

rzr, 2018-12-19:
Please check:
https://github.com/abandonware/noble/pull/1

rzr, 2019-05-05:
May this one be merged 1st:
https://github.com/noble/noble/pull/851

jackbridger, 2020-02-18:
Is there a recommended workaround to this? I am trying to run @abandonware/noble in a docker container? I receive this error:

`
/app/node_modules/@abandonware/noble/lib/hci-socket/hci.js:75
this._socket = new BluetoothHciSocket();
^

Error: EAFNOSUPPORT, Address family not supported by protocol
at new Hci (/app/node_modules/@abandonware/noble/lib/hci-socket/hci.js:75:18)
at new NobleBindings (/app/node_modules/@abandonware/noble/lib/hci-socket/bindings.js:28:15)
at Object. (/app/node_modules/@abandonware/noble/lib/hci-socket/bindings.js:511:18)
at Module._compile (internal/modules/cjs/loader.js:1151:30)
at Object.Module._extensions..js (internal/modules/cjs/loader.js:1171:10)
at Module.load (internal/modules/cjs/loader.js:1000:32)
at Function.Module._load (internal/modules/cjs/loader.js:899:14)
at Module.require (internal/modules/cjs/loader.js:1040:19)
at require (internal/modules/cjs/helpers.js:72:18)
at module.exports (/app/node_modules/@abandonware/noble/lib/resolve-bindings.js:13:12) {
errno: 97,
code: 'EAFNOSUPPORT',
syscall: 'socket'
}
`

I also posted this issue on abandonware https://github.com/abandonware/noble/issues/43