noble/noble #807: Stop readings after about 30 seconds.

TonySyl, 2018-06-13:
Hello to everyone. 

I have a glove with sensors and i want read the values from its sensors. The glove uses BLE for communication. 
I subscribe to the characteristic and then start reading the values. After about 30 seconds I stop receiving any new data and 10 seconds later, it disconnects from the glove. I have attached the output of the program and the result of the hcidump. 

Because the outputs where pretty long i have attached them in two txt files. 

I stop getting reading because of my code or because of the glove?

Thanks in advance.

[output.txt](https://github.com/noble/noble/files/2099104/output.txt)
[sniffer.txt](https://github.com/noble/noble/files/2099106/sniffer.txt)

Here is the code i am using:

    var noble = require('noble');

    noble.on('stateChange', function(state) {

        if (state === 'poweredOn') {
            noble.startScanning();
        } 
        else {
            noble.stopScanning();
       }

    });

    noble.on('discover',function(peripheral) {
    
        if(peripheral.advertisement.localName === "SensoDK2_L") {
            theData(peripheral);
        }

    });

    function theData(peripheral){

        peripheral.connect(function (error) {
        
            console.log(peripheral.state + ' to peripheral: ' + peripheral.uuid + ' ' + peripheral.advertisement.localName);
        
            peripheral.discoverServices(['a8bd67fb0c74471ebd1c9cec99925f02'], function (err, services) {
                var deviceServices = services[0];
                console.log("Services discovered");
            
                deviceServices.discoverCharacteristics(['a8bd00040c74471ebd1c9cec99925f02'], function (error, characteristics) {
                    var deviceCharacteristics = characteristics[0];
                    console.log("Characteristics discovered");
                    console.log(peripheral.state);
      
                    deviceCharacteristics.on('data', function (data, isNotification) {
                        console.log(peripheral.state);
                        console.log("The data: ", JSON.stringify(data));
                    });
                
                    deviceCharacteristics.subscribe(function (error) {
                        if (error) {
                            console.error('Error subscribing to echoCharacteristic');
                        } else {
                            console.log('Subscribed for echoCharacteristic notifications');
                        }
                    });
                });
            });
        });
    }

jrobeson, 2018-06-18:
what happens if you stop the scan before connecting?

TonySyl, 2018-06-18:
No change. I already tried to stop scanning before connecting.

jrobeson, 2018-08-03:
what OS is this on?

TonySyl, 2018-08-30:
@jrobeson It's on Windows 10.