noble/noble #808: Connection does not disconnect until device is found

jszersze, 2018-06-18:
Hello everyone, 

I'm having a problem with the connection not disconnecting until the device is powered on and found again. Let me explain my scenario:

1. I discover my device and hold the object in a list.
2. I unplug my device (to simulate user unplugging it or automatic powering off to save power).
3. I connect to the peripheral (`peripheral.connection(() => { ... })` - this will not return anything until the device is powered on again which is fine).
4. After some timeout (or device discovery coming around and determining that the device is lost), I attempt to disconnect the peripheral (`peripheral.disconnect(() => { ... })`

The problem is that 4. doesn't happen until the device is powered on again. This causes a problem because the user could select the device again to connect to it, and if they power it on again the connection seems to be compromised.

Any suggestions?

oumartraore, 2018-12-27:
Hello everyone,

I would like to restart this ticket, because I encounter exactly the same problem on Windows and Raspberry pi 3.

How to cancel a pending connection

kotkis1, 2019-09-10:
Hi everyone,

Same issue here. A pending connection blocks everything - no possible connection with other device until connection is established or node restarted. The major question is as stated by oumartraore:
How to cancel a pending connection?

Thanks in advance