noble/noble #817: Could not start scanning, state is unknown (not poweredOn)

m1dnight, 2018-07-18:
Hi,

I am running Noble version 1.8.1, on node.js version 8.11.3, on Mac OSX 10.13.4. 

I tried running your examples, but none of them seem to work. The adapter never switches to poweredOn. 

I have verified my bluetooth is turned on, and in OSX I can discover devices. `noble.state` returns `unknown`. 

Is this a known issue?

mattotodd, 2018-07-25:
Just out of curiosity,  do you only check the state once?  or are you watching for state changes?

```noble.on('stateChange', function(bluetoothHardwareState) {
      // check if it poweredOn
});```

alex20465, 2019-11-01:
Same issue here, in addition its not a logical solution to wait until the "stateChange" event, since the power can be from the beginning "poweredOn" (there is no state change).

I also noticed that only the "poweredOff" event is dispatched correctly, but the "poweredOn" NOT.

PopOS 19.10

Fabryz, 2020-07-16:
Does anyone have any tip on this? We're un Debian buster, Nodejs v12.18.2 on an Intel NUC device.
We start the application but the noble.state is always "unknown" despite the service is up and running. :(

Nyrox, 2021-07-21:
We are seeing this same issue on MacOS, multiple devices.
The extremely weird part is that this issue only shows up when attaching a state change listener.

We extended our internal bluetooth stack to monitor state changes like this 
`js
noble.on('stateChange',  (state: NobleState) => {
      this.state = state;
    });
`
Just registering this event listener causes the code to break. Without it the library runs just fine.