noble/noble #819: service.discoverCharacteristics is not a function

mattotodd, 2018-07-25:
I've been using this library for a while, and recently it starts throwing:

`service.discoverCharacteristics is not a function` when calling either:
`Peripheral.prototype.discoverSomeServicesAndCharacteristics` or
`Peripheral.prototype.discoverAllServicesAndCharacteristics`

I don't understand it, because if i add debug logging to to module, `discoverCharacteristics` is a function when i check the type. Any help would be greatly appreciated.

jrobeson, 2018-08-07:
you should be calling it like `peripheralInstance.discoverSomeServicesAndCharacteristics`

Zombitch, 2018-09-02:
Same issue here, did you find what gone wrong ?
I made a hot fix on peripheral.js at line 88 in order to add `if(service.discoverCharacteristics){` before calling `service.discoverCharacteristics(characteristicsUuids, function(error, characteristics)`

true-greg-anderson, 2018-12-16:
I ran into the same problem when trying to use noble in the same code as another frameworks.  That framework was manipulating object and array prototypes.  You can reproduce the problem in general by adding the following line (emulating what the other framework was doing) in the beginning of the test code: 
`Object.prototype.thenamedoesntmatter = function() {};`

Traced the issue to the same code as the above commenter.  In peripheral.js `Peripheral.prototype.discoverSomeServicesAndCharacteristics` line 85 needs to be corrected.

Current Line: 
`    for (var i in services) {`

Corrected Line:
`   for (var i=0;i<services.length;i++) {`

A similar correction should be applied to line 92.