noble/noble #820: Module not found: Can't resolve 'ws'

knandersen, 2018-07-30:
After installing noble and using it on latest Chrome on OSX, for some reason I get errors on resolving the websockets-module:

> Module not found: Can't resolve 'ws' in 'node_modules/noble/lib/websocket'

If I install websockets explicitely through `npm install ws` it works, so maybe there's a dependency missing?

jrobeson, 2018-07-30:
what errors do you get and while doing what? I'm using noble and i've never needed the ws module.

knandersen, 2018-07-30:
Sorry for not providing a thorough description. I get the error when running `npm start`: 

> Failed to compile.
> ./node_modules/noble/lib/websocket/bindings.js
> Module not found: Can't resolve 'ws' in 'nobletest/node_modules/noble/lib/websocket'

jrobeson, 2018-07-30:
and how did you install noble ? was `NODE_ENV` set to production?

knandersen, 2018-07-30:
noble was installed through npm as well, `npm install noble`. I use Create React App which has NODE_ENV set to development by default.

jrobeson, 2018-07-30:
try it outside of create-react-app.

knandersen, 2018-07-31:
I just tried doing some vanilla webpack, still get the same error.

index.js:

```
var noble = require('noble');
console.log("BLAH")
noble.startScanning(); // any service UUID, no duplicates
```

result:

```
MACBOOK:blah username$ ./node_modules/.bin/webpack --run-dev ./index.js -o bundle.js
ℹ ｢webpack｣: Starting Build
ℹ ｢webpack｣: Build Finished

webpack v4.16.3

05cc5b5499f43aa160bd
  size    name       module                       status
  109 B   index.js   ./index.js                   built
  489 B   global.js  (webpack)/buildin/global.js  built

  size    name       asset                        status
  431 kB  main       bundle.js                    emitted

  Δt 377ms (24 modules hidden)

./node_modules/noble/lib/websocket/bindings.js
  0:0  error  Module not found: Error: Can't resolve 'ws' in
              '/Users/username/Dev/blah/node_modules/noble/lib/websocket'

✖ 1 problem (1 error, 0 warnings)
```

jrobeson, 2018-07-31:
I should have been more specific, try it in something plain.

monteslu, 2018-07-31:
Try requiring it like this:
`var noble = require('noble/with-bindings')(require('noble/lib/webbluetooth/bindings'));`

knandersen, 2018-08-01:
@jrobeson it was probably me being slow. Thanks for explaining it to me.

@monteslu that works, thanks!

closed 2018-08-01 by knandersen