noble/noble #821: Write String data

rafifajar22, 2018-07-30:
Hey guys. can you give some example to send string data, i am very confused because it can only send 1 byte of data only.

Thank you

jrobeson, 2018-07-30:
No one can help if you don't show what you tried.

rafifajar22, 2018-07-31:
`var noble = require('noble');
var sleep = require('sleep');

var pspServiceUuid = 'd0cbda9a9a6949abb4d1a96d6fbd10c6';
var rxCharacteristUuid = 'd0cb6c999a6949abb4d1a96d6fbd10c6';

noble.on('stateChange', function(state) {
    if (state === 'poweredOn') {
        noble.startScanning([pspServiceUuid], false);
    } else {
        noble.stopScanning();
    }
});

noble.on('scanStart', function() {
  console.warn('Scan started');
});

noble.on('scanStop', function() {
  console.warn('Scan stopped');
});

noble.on('discover', function(peripheral) {
    if (peripheral.advertisement.localName !== "PSP-IoT") {
        return;
    }

    noble.stopScanning();

    peripheral.connect(function(err) {
        if (err) {
            console.error('Error connecting: ' + err);
            return;
        }
	
	peripheral.discoverServices([pspServiceUuid], function(err, services) {
      	  services.forEach(function(service) {

	console.log('found service:', service.uuid);
        console.log('Connected to ' + peripheral.advertisement.localName);
	
	service.discoverCharacteristics([rxCharacteristUuid], function(err, characteristics) {
          characteristics.forEach(function(characteristic) {
 		var commandWrite = characteristics[0];
        	console.log('discovered');

        // true if for write without response
        	commandWrite.write(new Buffer([0x0A,0x0A]), true, function(error) {
          	console.log('send');		
	
	sleep.sleep(3)
		commandWrite.write(new Buffer([0x0A,0x0A]), true, function(error) {
          	console.log('send');	
	sleep.sleep(3)
       peripheral.disconnect(function(error) {
       console.log('disconnected from peripheral: ' + peripheral.uuid);
	                    });
                        });
	            });
                });
            });      
        });
    });
});
});`

this my code i have try, whether can a newbuffer () handle string?, this is my first use of javascript