noble/noble #825: discover characteristics: works on Mac, empty array on Linux (Tessel)

gr2m, 2018-08-26:
Hey there,

I’ve a fairly simple code to connect to a bluetooth scale. It works reliably on my mac. But when I run the same code ton a [Tessel](https://tessel.io/) it works sometimes, but most of the time the characteristics array for the service is empty

```js
peripheral.discoverSomeServicesAndCharacteristics([WEIGHT_SERVICE_UUID], [], async (error, [service], characteristics) => {
  // characteristics is an empty array on tessel.io
})
```

The logs I see with `DEBUG=*`

```
Sun, 26 Aug 2018 04:37:08 GMT hci onSocketData: 024000070003000400020502
Sun, 26 Aug 2018 04:37:08 GMT hci       event type = 2
Sun, 26 Aug 2018 04:37:08 GMT hci onSocketData: 024020070003000400031700
Sun, 26 Aug 2018 04:37:08 GMT hci       event type = 2
Sun, 26 Aug 2018 04:37:08 GMT hci               cid = 4
Sun, 26 Aug 2018 04:37:08 GMT hci               handle = 64
Sun, 26 Aug 2018 04:37:08 GMT hci               data = 031700
Sun, 26 Aug 2018 04:37:08 GMT att d3:9e:60:f4:64:85: read: 031700
Sun, 26 Aug 2018 04:37:08 GMT att d3:9e:60:f4:64:85: new MTU is 23
Sun, 26 Aug 2018 04:37:08 GMT att d3:9e:60:f4:64:85: write: 100100ffff0028
Sun, 26 Aug 2018 04:37:08 GMT hci write acl data pkt - writing: 0240000b0007000400100100ffff0028
Sun, 26 Aug 2018 04:37:08 GMT hci onSocketData: 0240000b0007000400100100ffff0028
Sun, 26 Aug 2018 04:37:08 GMT hci       event type = 2
Sun, 26 Aug 2018 04:37:08 GMT hci onSocketData: 024020180014000400110601000700001808000b0001180c0016000a18
Sun, 26 Aug 2018 04:37:08 GMT hci       event type = 2
Sun, 26 Aug 2018 04:37:08 GMT hci               cid = 4
Sun, 26 Aug 2018 04:37:08 GMT hci               handle = 64
Sun, 26 Aug 2018 04:37:08 GMT hci               data = 110601000700001808000b0001180c0016000a18
Sun, 26 Aug 2018 04:37:08 GMT att d3:9e:60:f4:64:85: read: 110601000700001808000b0001180c0016000a18
Sun, 26 Aug 2018 04:37:08 GMT att d3:9e:60:f4:64:85: write: 101700ffff0028
Sun, 26 Aug 2018 04:37:08 GMT hci write acl data pkt - writing: 0240000b0007000400101700ffff0028
Sun, 26 Aug 2018 04:37:08 GMT hci onSocketData: 0240000b0007000400101700ffff0028
Sun, 26 Aug 2018 04:37:08 GMT hci       event type = 2
Sun, 26 Aug 2018 04:37:08 GMT hci onSocketData: 0240201a0016000400111417001c0023d1bcea5f782315deef121230150000
Sun, 26 Aug 2018 04:37:08 GMT hci       event type = 2
Sun, 26 Aug 2018 04:37:08 GMT hci               cid = 4
Sun, 26 Aug 2018 04:37:08 GMT hci               handle = 64
Sun, 26 Aug 2018 04:37:08 GMT hci               data = 111417001c0023d1bcea5f782315deef121230150000
Sun, 26 Aug 2018 04:37:08 GMT att d3:9e:60:f4:64:85: read: 111417001c0023d1bcea5f782315deef121230150000
Sun, 26 Aug 2018 04:37:08 GMT att d3:9e:60:f4:64:85: write: 101d00ffff0028
Sun, 26 Aug 2018 04:37:08 GMT hci write acl data pkt - writing: 0240000b0007000400101d00ffff0028
Sun, 26 Aug 2018 04:37:08 GMT hci onSocketData: 024020180014000400110601000700001808000b0001180c0016000a18
Sun, 26 Aug 2018 04:37:08 GMT hci       event type = 2
Sun, 26 Aug 2018 04:37:08 GMT hci               cid = 4
Sun, 26 Aug 2018 04:37:08 GMT hci               handle = 64
Sun, 26 Aug 2018 04:37:08 GMT hci               data = 110601000700001808000b0001180c0016000a18
Sun, 26 Aug 2018 04:37:08 GMT att d3:9e:60:f4:64:85: read: 110601000700001808000b0001180c0016000a18
Sun, 26 Aug 2018 04:37:08 GMT att d3:9e:60:f4:64:85: write: 101700ffff0028
Sun, 26 Aug 2018 04:37:08 GMT hci write acl data pkt - writing: 0240000b0007000400101700ffff0028
Sun, 26 Aug 2018 04:37:08 GMT hci onSocketData: 0240000b0007000400101d00ffff0028
Sun, 26 Aug 2018 04:37:08 GMT hci       event type = 2
Sun, 26 Aug 2018 04:37:08 GMT hci onSocketData: 0240000b0007000400100100ffff0128
Sun, 26 Aug 2018 04:37:08 GMT hci       event type = 2
Sun, 26 Aug 2018 04:37:08 GMT hci onSocketData: 0240201a0016000400111417001c0023d1bcea5f782315deef121230150000
Sun, 26 Aug 2018 04:37:08 GMT hci       event type = 2
Sun, 26 Aug 2018 04:37:08 GMT hci               cid = 4
Sun, 26 Aug 2018 04:37:08 GMT hci               handle = 64
Sun, 26 Aug 2018 04:37:08 GMT hci               data = 111417001c0023d1bcea5f782315deef121230150000
Sun, 26 Aug 2018 04:37:08 GMT att d3:9e:60:f4:64:85: read: 111417001c0023d1bcea5f782315deef121230150000
Sun, 26 Aug 2018 04:37:08 GMT att d3:9e:60:f4:64:85: write: 101d00ffff0028
Sun, 26 Aug 2018 04:37:08 GMT hci write acl data pkt - writing: 0240000b0007000400101d00ffff0028
Sun, 26 Aug 2018 04:37:08 GMT hci onSocketData: 0240000b0007000400101d00ffff0128
Sun, 26 Aug 2018 04:37:08 GMT hci       event type = 2
Sun, 26 Aug 2018 04:37:08 GMT hci onSocketData: 0240201a001600040011141d002f0032b2a2f753de1a9ceb47fd5e00002302
Sun, 26 Aug 2018 04:37:08 GMT hci       event type = 2
Sun, 26 Aug 2018 04:37:08 GMT hci               cid = 4
Sun, 26 Aug 2018 04:37:08 GMT hci               handle = 64
[... 232 lines cut ...]
Sun, 26 Aug 2018 04:33:30 GMT att d3:9e:60:f4:64:85: write: 082f002f000328
Sun, 26 Aug 2018 04:33:30 GMT hci write acl data pkt - writing: 0240000b0007000400082f002f000328
Sun, 26 Aug 2018 04:33:30 GMT hci onSocketData: 02402009000500040001082f000a
Sun, 26 Aug 2018 04:33:30 GMT hci       event type = 2
Sun, 26 Aug 2018 04:33:30 GMT hci               cid = 4
Sun, 26 Aug 2018 04:33:30 GMT hci               handle = 64
Sun, 26 Aug 2018 04:33:30 GMT hci               data = 01082f000a
Sun, 26 Aug 2018 04:33:30 GMT att d3:9e:60:f4:64:85: read: 01082f000a
Sun, 26 Aug 2018 04:33:32 GMT hci onSocketData: 01060403400013
Sun, 26 Aug 2018 04:33:32 GMT hci       event type = 1
Sun, 26 Aug 2018 04:33:32 GMT hci               cmd = 1030
Sun, 26 Aug 2018 04:33:32 GMT hci               data len = 3
Sun, 26 Aug 2018 04:33:32 GMT hci onSocketData: 040f0400010604
Sun, 26 Aug 2018 04:33:32 GMT hci       event type = 4
Sun, 26 Aug 2018 04:33:32 GMT hci       sub event type = 15
Sun, 26 Aug 2018 04:33:32 GMT hci               status = 0
Sun, 26 Aug 2018 04:33:32 GMT hci               cmd = 1030
Sun, 26 Aug 2018 04:33:32 GMT hci onSocketData: 04050400400016
Sun, 26 Aug 2018 04:33:32 GMT hci       event type = 4
Sun, 26 Aug 2018 04:33:32 GMT hci       sub event type = 5
Sun, 26 Aug 2018 04:33:32 GMT hci               handle = 64
Sun, 26 Aug 2018 04:33:32 GMT hci               reason = 22
```

I’d really appreciate any help 🙏 /cc @hipsterbrown

gr2m, 2018-08-26:
Here is the full log using `hcidump -t -x`

```
2018-08-26 08:20:44.250151 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2018-08-26 08:20:44.251171 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2018-08-26 08:20:44.253851 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2018-08-26 08:20:44.255131 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2018-08-26 08:20:44.255771 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2018-08-26 08:20:44.257151 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x1000
    LMP Version: 4.0 (0x6) LMP Subversion: 0x220e
    Manufacturer: Broadcom Corporation (15)
2018-08-26 08:20:44.257871 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00
2018-08-26 08:20:44.259131 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00
2018-08-26 08:20:44.259771 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
2018-08-26 08:20:44.261131 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00
2018-08-26 08:20:44.271791 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2018-08-26 08:20:44.273131 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 5C:F3:70:89:67:B5
2018-08-26 08:20:44.314693 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2018-08-26 08:20:44.316133 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2018-08-26 08:20:44.318493 < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
2018-08-26 08:20:44.320133 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2018-08-26 08:20:44.345454 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2018-08-26 08:20:44.347134 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2018-08-26 08:20:44.348794 < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
2018-08-26 08:20:44.350194 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2018-08-26 08:20:44.351674 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
2018-08-26 08:20:44.353174 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x00
2018-08-26 08:20:44.361095 > HCI Event: LE Meta Event (0x3e) plen 39
    LE Advertising Report
      ADV_NONCONN_IND - Non connectable undirected advertising (3)
[... 206 lines cut ...]
2018-08-26 08:20:49.798577 > ACL data: handle 64 flags 0x02 dlen 9
    ATT: Error (0x01)
      Error: Attribute not found (10)
      Read By Group req (0x10) on handle 0x0030
2018-08-26 08:20:49.806237 < ACL data: handle 64 flags 0x00 dlen 11
    ATT: Read By Type req (0x08)
      start 0x001d, end 0x002f
      type-uuid 0x2803
2018-08-26 08:20:49.814337 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
2018-08-26 08:20:49.828578 > ACL data: handle 64 flags 0x02 dlen 27
    ATT: Read By Type resp (0x09)
      length: 21
        handle 0x001e, value 0x12 0x1f 0x00 0x32 0xb2 0xa2 0xf7 0x53 0xde 0x1a 0x9c 0xeb 0x47 0xfd 0x5e 0x01 0x00 0x23 0x02
2018-08-26 08:20:49.834938 < HCI Command: Disconnect (0x01|0x0006) plen 3
    handle 64 reason 0x13
    Reason: Remote User Terminated Connection
2018-08-26 08:20:49.836338 > HCI Event: Command Status (0x0f) plen 4
    Disconnect (0x01|0x0006) status 0x00 ncmd 1
2018-08-26 08:20:49.844318 > HCI Event: Disconn Complete (0x05) plen 4
    status 0x00 handle 64 reason 0x16
    Reason: Connection Terminated by Local Host
```

The code looks for "Hatch Baby Grow", once it finds it, it stops scanning, connects to the peripheral and then looks for a specific characteristing using `discoverSomeServicesAndCharacteristics`. The `characteristics` array in the callback is empty.

gr2m, 2018-08-26:
When I tried to run it again, it gets stuck at `peripheral.discoverSomeServicesAndCharacteristics`, it never logs the "_X_ characteristics found for service 022300015efd47eb9c1ade53f7a2b232" message. 

<details>
  <summary>Here is the full log for that case</summary>

```
root@nursery:~# hcidump -t -x
HCI sniffer - Bluetooth packet analyzer ver 5.30
device: hci0 snap_len: 1500 filter: 0xffffffff
2018-08-26 17:47:58.531766 < HCI Command: Set Event Mask (0x03|0x0001) plen 8
    Mask: 0xfffffbff07f8bf3d
2018-08-26 17:47:58.533066 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Mask (0x03|0x0001) ncmd 1
    status 0x00
2018-08-26 17:47:58.536466 < HCI Command: LE Set Event Mask (0x08|0x0001) plen 8
    mask 0x1f00000000000000 (Reserved)
2018-08-26 17:47:58.538046 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Event Mask (0x08|0x0001) ncmd 1
    status 0x00
2018-08-26 17:47:58.540806 < HCI Command: Read Local Version Information (0x04|0x0001) plen 0
2018-08-26 17:47:58.542046 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x1000
    LMP Version: 4.0 (0x6) LMP Subversion: 0x220e
    Manufacturer: Broadcom Corporation (15)
2018-08-26 17:47:58.545406 < HCI Command: Write LE Host Supported (0x03|0x006d) plen 2
  01 00
2018-08-26 17:47:58.547046 > HCI Event: Command Complete (0x0e) plen 4
    Write LE Host Supported (0x03|0x006d) ncmd 1
    00
2018-08-26 17:47:58.548906 < HCI Command: Read LE Host Supported (0x03|0x006c) plen 0
2018-08-26 17:47:58.550046 > HCI Event: Command Complete (0x0e) plen 6
    Read LE Host Supported (0x03|0x006c) ncmd 1
    00 01 00
2018-08-26 17:47:58.551706 < HCI Command: Read BD ADDR (0x04|0x0009) plen 0
2018-08-26 17:47:58.553066 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 5C:F3:70:89:67:B5
2018-08-26 17:47:58.591688 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2018-08-26 17:47:58.593068 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2018-08-26 17:47:58.596228 < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
2018-08-26 17:47:58.597048 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2018-08-26 17:47:58.621729 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x00 (scanning disabled)
    filter duplicates 0x01 (enabled)
2018-08-26 17:47:58.623049 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Enable (0x08|0x000c) ncmd 1
    status 0x0c
    Error: Command Disallowed
2018-08-26 17:47:58.624349 < HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7
    type 0x01 (active)
    interval 10.000ms window 10.000ms
    own address: 0x00 (Public) policy: All
2018-08-26 17:47:58.625649 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Parameters (0x08|0x000b) ncmd 1
    status 0x00
2018-08-26 17:47:58.628330 < HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2
    value 0x01 (scanning enabled)
    filter duplicates 0x01 (enabled)
[... 58 lines cut ...]
      start 0x0001, end 0xffff
      type-uuid 0x2800
2018-08-26 17:47:59.062087 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
2018-08-26 17:47:59.076248 > ACL data: handle 64 flags 0x02 dlen 7
    ATT: MTU resp (0x03)
      server rx mtu 23
2018-08-26 17:47:59.077308 < ACL data: handle 64 flags 0x00 dlen 11
    ATT: Read By Group req (0x10)
      start 0x0001, end 0xffff
      type-uuid 0x2801
2018-08-26 17:47:59.083148 < ACL data: handle 64 flags 0x00 dlen 11
    ATT: Read By Group req (0x10)
      start 0x0001, end 0xffff
      type-uuid 0x2800
2018-08-26 17:47:59.092068 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 2
2018-08-26 17:47:59.106249 > ACL data: handle 64 flags 0x02 dlen 9
    ATT: Error (0x01)
      Error: Attribute not found (10)
      Read By Group req (0x10) on handle 0x0001
2018-08-26 17:47:59.339058 > HCI Event: Number of Completed Packets (0x13) plen 5
    handle 64 packets 1
```
</details>

coderbyheart, 2018-08-27:
> Question: can I subscribe to a characteristics right after connecting to the device if I know the service & characteristics UUIDs, without needing to discover the characteristic each time?

Yes. BLE discover is not neccessary, it only returns the available services, but does not set them in a "subscribeable" state.

coderbyheart, 2018-08-27:
Especially on small devices it's important to give the adapter some time. So try experimenting with pauses between operations.

gr2m, 2018-08-27:
Thanks @coderbyheart! I think I could find out all the UUIDs, the peripheral address and the start/end handles for the characteristic I’m interested in. But I couldn’t figure out how to directly subscribe to the weight characteristic using noble. If anyone could give me a hint, I’d appreciate it!

> Especially on small devices it's important to give the adapter some time. So try experimenting with pauses between operations.

Yes I did, but no luck unfortunately.

The only thing that has worked was not to connect to the scale with any device and turn of the tessel for a long while (e.g. overnight). Then it sometimes works once, then back to not working :(

reconbot, 2018-08-27:
What hardware are you using on the Tessel for bluetooth? The little bluetooth dongles are not all created equal. I know @monteslu has been around the block on this.

Delays aside, IIRC I've had success in discovering multiple times until I see what I want. 

As for reading a characteristic directly. It looks like you have [two of the three](https://github.com/noble/noble/blob/master/lib/characteristic.js#L8-L24) required pieces of data as constants but I couldn't tell you if you should construct this yourself.

gr2m, 2018-08-28:
> What hardware are you using on the Tessel for bluetooth? The little bluetooth dongles are not all created equal. I know @monteslu has been around the block on this.

I use the Pluggable linked at https://tessel.io/modules#noble

> I've had success in discovering multiple times until I see what I want

I tried that, too. No luck, it was an endless loop.

> It looks like you have two of the three required pieces of data as constants

I was able to construct it myself using the peripheral address, the UUIDs for service & characteristic and start/end handle for the service. But no `data` events were emitted, I couldn’t figure out why. I had to hack the noble code at several places. I hope there is an easier way

homedad, 2018-09-22:
I had the same problem using a mac to read characteristics from my Raspberry Pi. A delay between discover services and discover characteristics solved the problem