noble/noble #828: Use official CoreBluetooth API

geovie, 2018-09-18:
Hi,
I rewrote the mac bindings to use native modules and the official CoreBluetooth API.

https://github.com/Timeular/noble-mac

The bindings provide the same functionality as the current mac bindings in this repository.

Are the noble developer interested in merging this in? The main benefit would be that the XPC protocol does not have to be reverse engineered every new MacOS release https://github.com/noble/noble/pull/727 https://github.com/noble/noble/pull/804

Kudos to [Timeular](https://timeular.com) for sponsoring my work on this.

drandreas, 2018-09-23:
Thank you for your effort!

I've tried to verify your implementation against macOS Mojave. But currently all my node-gyp builds fail. I assume the root cause is the disabled support for 32bit builds in Mojave's SDK.

I will investigate further.

readeral, 2018-09-26:
Hopefully this can be tightened up and merged! Will go a long way to solving some other open issues.

drandreas, 2018-09-27:
Mojave is now public. Did any one else succeed in getting it to work?

geovie, 2018-09-28:
@drandreas Can you try it again with the latest master? I've added node-pre-gyp so it should work for you

drandreas, 2018-10-01:
Sorry for the delay.

It works perfectly.

geovie, 2018-11-09:
FYI I opened PR #840 which integrates the changes from [noble-mac](https://github.com/Timeular/noble-mac)