noble/noble #831: Index out of range error in processLeAdvertisingReport

AlCalzone, 2018-09-22:
```
RangeError: Index out of range
at BluetoothHciSocket.emit (events.js:211:7)
at emitOne (events.js:116:13)
at Hci.onSocketData (/opt/iobroker/node_modules/noble/lib/hci-socket/hci.js:479:12)
at Hci.processLeMetaEvent (/opt/iobroker/node_modules/noble/lib/hci-socket/hci.js:608:10)
at Hci.processLeAdvertisingReport (/opt/iobroker/node_modules/noble/lib/hci-socket/hci.js:643:21)
at Buffer.readInt8 (buffer.js:1103:5)
at checkOffset (buffer.js:977:11)
```
I believe the buffer's length should be checked before reading from it in order to avoid crashing on invalid packets.

svogl, 2019-04-24:
we hade similar problems and added two try-catch regions around the affected hci and gap code; you might want to give this a try:
https://github.com/voxel-dot-at/noble/commit/75a34d989bd0557bf3239e1cdf2632bf2b4e6d8c

johngo7470, 2019-11-10:
svogl - Your updates to handle invalid packets, are the cat's meow - thanks!