noble/noble #834: macOS Mojave Not Discovering Peripherals

jaret32, 2018-10-22:
I just upgraded to macOS Mojave and am running version 10.14.  Everything was working before upgrading, but now the peripherals are not being discovered.

tagrim, 2018-10-23:
Same for me (Mojave 10.14, MacBook Pro (Retina, 15-inch, Mid 2015))

tagrim, 2018-10-23:
figured out. some xpc events changed. hope author will fix this, or ill suggest a PR

monteslu, 2018-10-23:
There's an open PR here:  https://github.com/noble/noble/pull/804  
More testing and code review on that would be appreciated.

moki298, 2018-10-25:
Happened the same with me too, don't see the peripherals after the Mojave update.

webartistse, 2018-10-29:
Same here. Does not found anything at all. Mojave 10.14 + Xcode 10.0.

monteslu, 2018-10-29:
@moki298 @webartistse @akinyema take a look at the PR I mentioned and test, please.

akinyema, 2018-10-29:
@monteslu Hi, i don't really understand how to test it, which file do i need to put that code/change? Thanks

jaret32, 2018-10-29:
@monteslu I've tried using the [Timeular/noble-mac](https://github.com/Timeular/noble-mac), but it is still not discovering the peripheral that I am trying to connect to.  It does discover other peripherals but not the one I need.

tagrim, 2018-10-29:
> @monteslu I've tried using [Timeular/noble-mac](https://github.com/Timeular/noble-mac), but it is still not discovering the peripheral that I am trying to connect to. It does discover other peripherals but not the one I need.

did they connectable? check ur configs. noblemac used patch fixes that working. or just to dig deeper (enable debug npm pack for noble and check what devices it discover. possibly it can be device issue that not broadcasting required info/perms)

jaret32, 2018-10-29:
Is it possible that it's not discoverable with noble-mac even though it was discoverable with noble before MacOS Mojave? What should I check for as far as broadcasting goes? I can see my peripheral through the Light Blue Explorer app on my iPhone 7.

tagrim, 2018-10-29:
> Is it possible that it's not discoverable with noble-mac even though it was discoverable with noble before MacOS Mojave? What should I check for as far as broadcasting goes? I can see my peripheral through the Light Blue Explorer app on my iPhone 7.

if u want, u can get patch changes and use it by own in node_modules, adding some files / replacing some code. 

also try to clean node_modules and npm cache. coz changes only in pointers. so i can bet dat issue in caches

tagrim, 2018-10-29:
also, u can directly checkout the commit, setting url in package.json file

webartistse, 2018-10-29:
@monteslu I installed noble-mac and it worked as expected.

tagrim, 2018-10-29:
@monteslu gonna run some tests tomorrow evening, but in common seems legit

jaret32, 2018-10-29:
@webartistse What device are you connecting to? I've cleaned the cache and re-installed all modules, but the peripheral still isn't being found.  It may be an issue with the device, but if so, then noble-mac is no longer compatible with a subset of devices that worked fine with noble.

jaret32, 2018-10-29:
I have it in my package.json file as a dependency.  This is what I have, "noble-mac": "git://github.com/Timeular/noble-mac.git"

tagrim, 2018-10-29:
@jaret32 what kind of device u trying to connect? and rly try checkout exact commit

jaret32, 2018-10-29:
@tagrim https://learn.adafruit.com/adafruit-feather-32u4-bluefruit-le/overview

webartistse, 2018-10-29:
@jaret32 I havn't done more than finding peripherals atm. With noble I didn't find a single one, after installed noble-mac I can find around 30 devices.

jaret32, 2018-10-29:
Yes, I find devices as well now, just not the one I need haha.

webartistse, 2018-10-29:
@jaret32 I may have the same problem then. Because I tried to find some ibeacons. I didn't find them right away. I will try to change the batteries because they don't have any led to show if they are on or off.

jaret32, 2018-10-29:
Okay, let me know.  Thanks!

jaret32, 2018-10-31:
This seems to work so far https://github.com/Timeular/noble-mac/issues/7

charliegerard, 2019-04-30:
@zach-luman I installed noble-mac from [here](https://github.com/Timeular/noble-mac) and it's working really well for me!

marksyzm, 2019-04-30:
@charliegerard you might like to use the @abandonware/noble account then as it's started being maintained more over there 🙂

ideacco, 2019-10-11:
Now on October 11, 2019, I used Mojave 10.14.6, running the code found in the sample documentation, and still have no output, so this problem has not been solved yet?

slayerjain, 2019-11-17:
I'm also wondering about compatibility with recent versions of macOS. I'm on 10.14

rzr, 2020-09-06:
Relate-to: https://github.com/noble/noble/issues/923#issuecomment-687851954