noble/noble #838: Reconnecting peripheral

markdyousef, 2018-10-29:
Trying to implement a method to automatically reconnect to disconnected peripherals, I ran into the following issue:
Listening to the "disconnect" event emitted once the peripheral has lost connection. I start a new scan and try to re-discover the lost peripheral (using uuid). Once discovered, I establish a connection which works fine (the first time). The problem occurs when I connect and disconnect multiple times, then `noble` seems to connect to multiple instances of the peripheral.

I recently upgraded to Mojave, using [Timeular's solution](https://github.com/Timeular/noble/commit/e930579c46754565442f1154003fe6e6a1203ee1#diff-2c29a693d41bf015ac8fec78b8d8d996) to fix some issues with discover. I'm not sure if the issue is in the `noble 1.9.1` module or the added bindings.

vincenthure, 2019-01-15:
I am using noble with a raspberry and nodejs 8
I get the same problem.
When I connect the first time everything work fine.
After "disconnect" and scanning I get my device againt, I can write to it, but I can't read anything because callback don't work
the only solution is to quit noble and start it again to read information on the device.
My second device is also a raspberry PI3, with bleno

paulampkelly, 2019-03-01:
I have a similar issue using a BBC Micro:bit as the peripheral and a rasbberry as the central running nodejs8. 
When I scan the first time the peripheral appears and I can connect to it and interrogate it for its services and characteristics.
After calling the peripheral's disconnect function and re-scanning the peripheral does not appear again so I cannot reconnect to it. The disconnect connect callback is executed so I have definitely disconnected. Shouldn't that peripheral then appear again on rescanning?