noble/noble #842: Raspberry Pi 3/raspbian stretch 9: peripherals are found - just not by Noble

fabdrol, 2018-11-16:
I'm using Noble on a RPi3 with this software: https://github.com/decipherindustries/signalk-calypso-ultrasonic (use `test/standalone.js` to run without SK server). The peripheral I'm looking for, named ULTRASONIC, is never found - whereas I _can_ see it when I do a manual scan.

Any ideas?

shadycuz, 2018-11-24:
@fabdrol I have a similar issue, I'm on raspbian on a Zero W. I can see devices using the exampe code, but not my phone. I can see when I use the built-in bluethoothctl cli tool. I can also see it when I use python libraries to scan, but not from noble =/

Pythonaire, 2019-02-19:
same here, but more strange. Device found(after some "uh, oh no command" - messages), but receiving values is more coincidence. Maybe, node version conflict? Last usable node.js version on Zero W might be 9.9, but the noble travis file tells node 8 ... strange.