noble/noble #847: Trapping "No compatible USB Bluetooth 4.0 device" error?

gfwilliams, 2018-12-06:
Hi!

I have a project that uses Noble as one of a few different connection options (BLE, USB Serial, TCP/IP) - however I'm finding that on Windows 10 I get `No compatible USB Bluetooth 4.0 device` errors on some devices which closes my application.

That's fine - I know some devices won't have BLE dongles that work with Noble. However I'd like to be able to trap the error gracefully and carry on without Noble regardless.

On Linux, I get `stateChange -> poweredOff` without any devices, so that's easy.

I have tried:

```
  var noble = undefined;
  try {
    noble = require('noble');
  } catch (e) {
   ...
  }
```

But that doesn't seem to be able to trap the error that occurs on Windows 10 as it appears it's coming from a callback that's been scheduled at some later point.

```
Error: No compatible USB Bluetooth 4.0 device found!
at BluetoothHciSocket.bindUser (C:\Users\name\AppData\Roaming\npm\node_modules\espruino\node_modules\bluetooth-hci-socket\lib\usb.js:70:11)
at BluetoothHciSocket.bindRaw (C:\Users\name\AppData\Roaming\npm\node_modules\espruino\node_modules\bluetooth-hci-socket\lib\usb.js:28:8)
at Hci.init (C:\Users\name\AppData\Roaming\npm\node_modules\espruino\node_modules\noble\lib\hci-socket\hci.js:101:35)
at NobleBindings.init (C:\Users\name\AppData\Roaming\npm\node_modules\espruino\node_modules\noble\lib\hci-socket\bindings.js:82:13)
at Noble.<anonymous> (C:\Users\name\AppData\Roaming\npm\node_modules\espruino\node_modules\noble\lib\noble.js:57:24)
at _combinedTickCallback (internal/process/next_tick.js:131:7)
at process._tickCallback (internal/process/next_tick.js:180:9)
at Function.Module.runMain (module.js:695:11)
at startup (bootstrap_node.js:188:16)
at bootstrap_node.js:609:3
```

Is there any way I can trap it nicely? About the only thing I can think right now is `uncaughtException` but that seems like overkill.

GavinRoman, 2019-08-02:
I had the same problem, and then I find a solution. There are two steps: 
1.npm install usb
2.try {
            var usb = require('usb');
            usb.open();
           ...
        }
        catch (error) {
           ...
        }

GavinRoman, 2019-08-14:
I've a method and it works well.

1. install dependecy :
 yarn add bluetooth-hci-socket
2. show the code:

 try {
            //check support
            var BluetoothHciSocket = require('bluetooth-hci-socket');
            var bluetoothHciSocket = new BluetoothHciSocket();
            bluetoothHciSocket.bindUser();
} catch (error) {
           
       }

gfwilliams, 2019-08-27:
Thanks - however is it possible that having previously bound the socket will then interfere with `noble`?

GavinRoman, 2019-09-12:
It won't interfere with noble. I has applied the code in my project. It works very well

gfwilliams, 2019-09-12:
Thanks - I'll see about adding this to https://github.com/espruino/EspruinoTools for Windows - however IMO this should still be handled in noble.

paramsiddharth, 2022-07-22:
@gfwilliams Do we have any updates? I think this might be related to #133.

gfwilliams, 2022-07-25:
All I currently do to work around it is this: https://github.com/espruino/EspruinoTools/blob/974b572217b724d4c29c9ca9e3c86385e9fcd765/core/serial_noble.js#L50-L71

So I just install an uncaughtException handler. Also I'm now using https://github.com/abandonware/noble as it's a bit better maintained.

Closing this as there's a workaround

closed 2022-07-25 by gfwilliams