noble/noble #850: Cannot install noble on RPI3B, RPI 3B+ (Latest Raspbian) using nodeJS 10

Theofilos-Chamalis, 2018-12-18:
I am not able to install (npm i) noble on RPI3B, RPI 3B+ (Latest Raspbian) using nodeJS 10.

The output that I am present is the following:

> node-pre-gyp install --fallback-to-build

node-pre-gyp WARN Using request for node-pre-gyp https download 
node-pre-gyp WARN Tried to download(404): https://github.com/tessel/node-usb/releases/download/1.5.0/usb_bindings-v1.5.0-node-v64-linux-arm.tar.gz 
node-pre-gyp WARN Pre-built binaries not found for usb@1.5.0 and node@10.14.2 (node-v64 ABI, glibc) (falling back to source compile with node-gyp) 
make: Entering directory '/home/pi/VieConnect/Projects/Secco/Scanner/node_modules/usb/build'
  CC(target) Release/obj.target/libusb/libusb/libusb/core.o
  CC(target) Release/obj.target/libusb/libusb/libusb/descriptor.o
  CC(target) Release/obj.target/libusb/libusb/libusb/hotplug.o
  CC(target) Release/obj.target/libusb/libusb/libusb/io.o
  CC(target) Release/obj.target/libusb/libusb/libusb/strerror.o
  CC(target) Release/obj.target/libusb/libusb/libusb/sync.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/poll_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/threads_posix.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_usbfs.o
  CC(target) Release/obj.target/libusb/libusb/libusb/os/linux_udev.o
  AR(target) Release/obj.target/usb.a
  COPY Release/usb.a
  CXX(target) Release/obj.target/usb_bindings/src/node_usb.o
../src/node_usb.cc: In function ‘Nan::NAN_METHOD_RETURN_TYPE SetDebugLevel(Nan::NAN_METHOD_ARGS_TYPE)’:
../src/node_usb.cc:99:73: warning: ‘uint32_t v8::Value::Uint32Value() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
  if (info.Length() != 1 || !info[0]->IsUint32() || info[0]->Uint32Value() > 4) {
                                                                         ^
In file included from /home/pi/.node-gyp/10.14.2/include/node/v8.h:26:0,
                 from ../src/node_usb.h:12,
                 from ../src/node_usb.cc:1:
/home/pi/.node-gyp/10.14.2/include/node/v8.h:2477:47: note: declared here
   V8_DEPRECATED("Use maybe version", uint32_t Uint32Value() const);
                                               ^
/home/pi/.node-gyp/10.14.2/include/node/v8config.h:324:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/node_usb.cc:103:53: warning: ‘uint32_t v8::Value::Uint32Value() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
  libusb_set_debug(usb_context, info[0]->Uint32Value());
                                                     ^
In file included from /home/pi/.node-gyp/10.14.2/include/node/v8.h:26:0,
                 from ../src/node_usb.h:12,
                 from ../src/node_usb.cc:1:
/home/pi/.node-gyp/10.14.2/include/node/v8.h:2477:47: note: declared here
   V8_DEPRECATED("Use maybe version", uint32_t Uint32Value() const);
                                               ^
/home/pi/.node-gyp/10.14.2/include/node/v8config.h:324:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/node_usb.cc: In function ‘void handleHotplug(std::pair<libusb_device*, libusb_hotplug_event>)’:
../src/node_usb.cc:151:58: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)’ is deprecated [-Wdeprecated-declarations]
  Nan::MakeCallback(Nan::New(hotplugThis), "emit", 2, argv);
                                                          ^
In file included from ../src/helpers.h:3:0,
                 from ../src/node_usb.h:21,
                 from ../src/node_usb.cc:1:
../../nan/nan.h:1001:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
../src/node_usb.cc: In function ‘v8::Local<v8::Value> libusbException(int)’:
../src/node_usb.cc:301:14: warning: ‘v8::Local<v8::Object> v8::Value::ToObject() const’ is deprecated: Use maybe version [-Wdeprecated-declarations]
  e->ToObject()->Set(Nan::New<String>("errno").ToLocalChecked(), Nan::New<Integer>(errorno));
              ^
In file included from ../src/node_usb.h:12:0,
                 from ../src/node_usb.cc:1:
/home/pi/.node-gyp/10.14.2/include/node/v8.h:10046:15: note: declared here
 Local<Object> Value::ToObject() const {
               ^~~~~
  CXX(target) Release/obj.target/usb_bindings/src/device.o
../src/device.cc: In static member function ‘static void Req::default_after(uv_work_t*)’:
../src/device.cc:237:64: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’ is deprecated [-Wdeprecated-declarations]
[... 102 lines cut ...]
gyp ERR! stack     at ChildProcess.onExit (/home/pi/.nvm/versions/node/v10.14.2/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:240:12)
gyp ERR! System Linux 4.14.89-v7+
gyp ERR! command "/home/pi/.nvm/versions/node/v10.14.2/bin/node" "/home/pi/.nvm/versions/node/v10.14.2/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/VieConnect/Projects/Secco/Scanner/node_modules/bluetooth-hci-socket
gyp ERR! node -v v10.14.2
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

added 268 packages from 737 contributors and audited 1174 packages in 94.47s
found 1 low severity vulnerability
  run `npm audit fix` to fix them, or `npm audit` for details

I have also tried the fix regarding adding the xpc-connection dependency (sandeepmistry/node-xpc-connection#pull/26/head) in the package.json, but it is failing bc it requires macOS instead of linux.

Any suggestion (except reverting to older NodeJS versions like NodeJS 8) would be much appreciated.

Thank you

rzr, 2018-12-19:
yes this is a known issue I am solving it with:
https://github.com/abandonware/noble/pull/1

Feedback from maintainers is welcome, but PR queue is growing...

vincenthure, 2019-01-15:
I have the same problem
Solution  : remove Nodejs 10 and install Nodejs 8
now it's working I can install "bluetooth-hci-socket"
If it's still not working start from scratch with rasbian and install Nodejs 8

Whoweez, 2019-04-19:
I have the same problem running:
node v11.14.0
npm v6.7.0
Raspberry Pi 3B+

**** same as above: bla bla bla and then error:

> bluetooth-hci-socket@0.5.1 install /home/pi/node_modules/bluetooth-hci-socket
> node-gyp rebuild

make: Entering directory '/home/pi/node_modules/bluetooth-hci-socket/build'
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
../src/BluetoothHciSocket.cpp: In member function 'void BluetoothHciSocket::poll()':
../src/BluetoothHciSocket.cpp:251:95: warning: 'v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)' is deprecated [-Wdeprecated-declarations]
     Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
                                                                                               ^
In file included from ../src/BluetoothHciSocket.cpp:8:0:
../../nan/nan.h:980:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
../src/BluetoothHciSocket.cpp: In member function 'void BluetoothHciSocket::emitErrnoError()':
../src/BluetoothHciSocket.cpp:275:72: error: no matching function for call to 'v8::Function::NewInstance(int, v8::Local<v8::Value> [1])'
   Local<Value> error = errorConstructor->NewInstance(1, constructorArgs);
                                                                        ^
In file included from /home/pi/.node-gyp/11.14.0/include/node/node.h:63:0,
                 from /home/pi/.node-gyp/11.14.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/home/pi/.node-gyp/11.14.0/include/node/v8.h:4105:44: note: candidate: v8::MaybeLocal<v8::Object> v8::Function::NewInstance(v8::Local<v8::Context>, int, v8::Local<v8::Value>*) const
   V8_WARN_UNUSED_RESULT MaybeLocal<Object> NewInstance(
                                            ^~~~~~~~~~~
/home/pi/.node-gyp/11.14.0/include/node/v8.h:4105:44: note:   candidate expects 3 arguments, 2 provided
/home/pi/.node-gyp/11.14.0/include/node/v8.h:4108:44: note: candidate: v8::MaybeLocal<v8::Object> v8::Function::NewInstance(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT MaybeLocal<Object> NewInstance(
                                            ^~~~~~~~~~~
/home/pi/.node-gyp/11.14.0/include/node/v8.h:4108:44: note:   candidate expects 1 argument, 2 provided
../src/BluetoothHciSocket.cpp:282:93: warning: 'v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)' is deprecated [-Wdeprecated-declarations]
   Nan::MakeCallback(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
                                                                                             ^
In file included from ../src/BluetoothHciSocket.cpp:8:0:
../../nan/nan.h:980:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
../src/BluetoothHciSocket.cpp: In static member function 'static Nan::NAN_METHOD_RETURN_TYPE BluetoothHciSocket::BindRaw(Nan::NAN_METHOD_ARGS_TYPE)':
../src/BluetoothHciSocket.cpp:395:34: warning: 'int64_t v8::Value::IntegerValue() const' is deprecated: Use maybe version [-Wdeprecated-declarations]
       devId = arg0->IntegerValue();
                                  ^
In file included from /home/pi/.node-gyp/11.14.0/include/node/v8.h:26:0,
                 from /home/pi/.node-gyp/11.14.0/include/node/node.h:63,
                 from /home/pi/.node-gyp/11.14.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/home/pi/.node-gyp/11.14.0/include/node/v8.h:2570:46: note: declared here
   V8_DEPRECATED("Use maybe version", int64_t IntegerValue() const);
                                              ^
/home/pi/.node-gyp/11.14.0/include/node/v8config.h:326:3: note: in definition of macro 'V8_DEPRECATED'
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/BluetoothHciSocket.cpp: In static member function 'static Nan::NAN_METHOD_RETURN_TYPE BluetoothHciSocket::BindUser(Nan::NAN_METHOD_ARGS_TYPE)':
../src/BluetoothHciSocket.cpp:417:34: warning: 'int64_t v8::Value::IntegerValue() const' is deprecated: Use maybe version [-Wdeprecated-declarations]
       devId = arg0->IntegerValue();
                                  ^
In file included from /home/pi/.node-gyp/11.14.0/include/node/v8.h:26:0,
                 from /home/pi/.node-gyp/11.14.0/include/node/node.h:63,
                 from /home/pi/.node-gyp/11.14.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/home/pi/.node-gyp/11.14.0/include/node/v8.h:2570:46: note: declared here
   V8_DEPRECATED("Use maybe version", int64_t IntegerValue() const);
                                              ^
/home/pi/.node-gyp/11.14.0/include/node/v8config.h:326:3: note: in definition of macro 'V8_DEPRECATED'
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
binding.target.mk:99: recipe for target 'Release/obj.target/binding/src/BluetoothHciSocket.o' failed
make: *** [Release/obj.target/binding/src/BluetoothHciSocket.o] Error 1
make: Leaving directory '/home/pi/node_modules/bluetooth-hci-socket/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
[... cut ...]
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@0.1.4 (node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

Any help around this would help!

Whoweez, 2019-04-19:
> I have the same problem
> Solution : remove Nodejs 10 and install Nodejs 8
> now it's working I can install "bluetooth-hci-socket"
> If it's still not working start from scratch with rasbian and install Nodejs 8

I think I tried 8.1.16 earlier and that didn't work either... Can you confirm if it works on Rpi 3B+?

dsteinman, 2019-05-31:
Use this fork: https://github.com/abandonware/noble

rzr, 2019-06-01:
Relate-to: https://github.com/abandonware/noble/issues/2