noble/noble #856: Cannot connect again after disconnect

vincenthure, 2019-01-17:
I use two RPI3, one with noble and one with bleno.
I start RPI noble and RPI bleno  -> It's OK connection work
I stop RPI bleno -> It's OK RPI noble tell me "disconnect"
I start again RPI Bleno -> RPI noble tell me that a device is connected
And now I can write to it, but I can't read from it because the callback don't work.

I am blocked since many days with this problem, I read many thing on internet but I don't find any solution and I don't understand why I can't connect twice the same device!
Any suggestion will bee welcome

var noble = require('noble');

noble.on('stateChange',
	function(state)
		{
		console.log("state Change");
		if( state === 'poweredOn')
			{
			console.log("start scanning");
			noble.startScanning([SERVICE_UUID],false);
			}
		else
			{
			noble.stopScanning();
			}
		});

noble.on('discover',function(peripheral)
		{
		console.log("Discover");
		noble.stopScanning();
					
		peripheral.once('disconnect', function() 
			{
			console.log('Deconnection');
			noble.startScanning([SERVICE_UUID],false);
			});

		peripheral.connect( function() 
			{
			console.log('Connection');
			peripheral.discoverSomeServicesAndCharacteristics([SERVICE_UUID], [CP_UUID], 
				function(error, services, characteristics) 
					{
                                         console.log('new connection')
					// Do something
					})
			})	
		});

abhishekkumardwivedi, 2019-02-22:
Can you please check #864 if something helps and useful.

justAnotherDev, 2023-02-16:
[This](https://github.com/noble/noble/issues/692#issuecomment-335408717) was the answer for me.

rzr, 2024-02-06:
May you forward the issue to @abandoware/noble until upstream gives feedback at:

https://github.com/abandonware/noble/issues/2