noble/noble #859: Bluetooth 5.0 devices not recognized

tomasero, 2019-02-06:
Hello,

I am trying to receive data from an NRF52 device, which implements BLE 5. I am able to interface with this device from an iPhone, but unable to from a Mac. However, I am able to recognize and receive data from a BLE 4 device. Any insights on how to fix this?

Thanks!
Tomás

dsteinman, 2019-05-27:
I would try on this fork:

https://github.com/abandonware/noble

I have an NRF52832 from SparkFun and that worked, and I'll soon be trying with one of the newer 840 boards as well.

zkycloudz, 2019-06-13:
Hello, I am also try to work with nRF52840 from this link [Buetooth nRF52840](https://www.aliexpress.com/item/Nordic-nRF52840-BLE-dongle-bluetooth-4-0-bluetooth-5-0-for-computer/32951812498.html?spm=2114.search0104.3.29.7d0c110bxAFi1V&ws_ab_test=searchweb0_0,searchweb201602_8_10065_10068_319_10059_10884_317_10887_10696_321_322_10084_453_10083_454_10103_10618_10307_537_536,searchweb201603_53,ppcSwitch_0&algo_expid=5a20d96c-4f2e-4782-b71f-d4f38a26b7e5-4&algo_pvid=5a20d96c-4f2e-4782-b71f-d4f38a26b7e5)

The problem is it can not be detected from Noble library. I already install the driver using zadig tool, but still can't make it work.

dsteinman, 2019-06-13:
I have both of these products and noble is working very well on them using that abandonware fork.

https://learn.adafruit.com/introducing-the-adafruit-nrf52840-feather/overview

https://www.sparkfun.com/products/15025

zkycloudz, 2019-06-14:
Is there any issue you found in implementing the abandonware fork? I saw there is some issue which Manufacturer Data is Lost, how can you explain this? Is there any strange behaviour other the issue I mention?

dsteinman, 2019-06-14:
Works perfectly for me on both OSX and Raspbian.  Yes manufacturerData is not there by default, you'd have to pull in that commit.