noble/noble #862: Service shows up in gatttool but not with noble

gontranpic, 2019-02-14:
Hi, 

I am using a simple peripheral discovery code (see below) using noble and I am facing the following issue: one of the available services is showing up when I am using gatttool to connect to the BLE device but not showing up when using noble. The service is used for FOTA updates (`f000ffc0-0451-4000-b000-000000000000`)

I find it pretty strange that the service is showing up in gatttool but not through noble. 
Any help would be greatly appreciated!

**Output gatttool**   `sudo gatttool -b 54:6C:0E:E2:A8:49 -I `
```
attr handle 0x0001, end grp handle: 0x0007 uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle 0x0008, end grp handle: 0x0008 uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle 0x0009, end grp handle: 0x0019 uuid: 0000180a-0000-1000-8000-624f902fa84f
attr handle 0x001a, end grp handle: 0x0039 uuid: 5d5b1447-f938-4e72-ba34-e55716a49afb
attr handle 0x003a, end grp handle: 0x0042 uuid: f242e492-88b7-462e-8889-00805f9b34fb
attr handle 0x0043, end grp handle: 0xffff uuid: f000ffc0-0451-4000-b000-000000000000
```

**Output Node js code**
```
peripheral with ID 546c0ee2b7f0  rssi = -60    named    TestDevice  found
0   1800
1   1801
2   180a
3   5d5b1447f9384e72ba34624f902fa83f
4   f242e49288b7462e8889e55716a49afb
```

**Node js code**
```
noble.on('stateChange',
function(state) {
if (state === 'poweredOn') {
    noble.startScanning();
  } else {
    noble.stopScanning();
  }
});
noble.on('discover', function(peripheral) {
  if (peripheral.advertisement.localName && (peripheral.advertisement.localName.includes(TDL_Name) || peripheral.address === peripheralIdOrAddress)) {
    noble.stopScanning();
    console.log('peripheral with ID ' + peripheral.id + ' rssi = '+peripheral.rssi+'   named   '+peripheral.advertisement.localName+' found');
    process(peripheral);
  }
});
function process(peripheral) {
  peripheral.on('connect',function(){
        this.discoverServices();
      });
  peripheral.on('disconnect',function(){
        console.log('disconnected');
      });
  peripheral.on('servicesDiscover',function(services){
        for (var i in services){console.log(i+"  "+services[i].uuid);}
 });
  peripheral.connect();
}
```

abhishekkumardwivedi, 2019-02-22:
Can you check #864 , not sure but maybe connect would be failing for you without acknowledging to the application. You can try below code with having pull request #865 for using reset command.
```
noble.reset();
sleep.msleep(300);
peripheral.connect(..);
```

gontranpic, 2019-02-22:
Hi Abhishek,

Thanks for the feedback. I have further investigated the issue with "sudo btmon" and it looks like the Handle for the service I am looking for is there also when using noble but there is an authentification failure when the pairing takes place.

The pairing also fails in gatttool actually so this comes from the device. I have contacted the dev team for more support.

We can close the issue since the original question is solved

closed 2019-02-22 by gontranpic