noble/noble #874: npm install fails - mojave

Narayan-Gopalappa, 2019-03-21:
I have XCode installed.

> xpc-connection@0.1.4 install /Users/narayan.gopalappa/Desktop/Me/Git/TestBLE/node_modules/xpc-connection
> node-gyp rebuild

xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instan
ce

xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instan
ce

Traceback (most recent call last):
  File "/Users/narayan.gopalappa/.nvm/versions/node/v6.0.0/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py", line 16, in <module>
    sys.exit(gyp.script_main())
  File "/Users/narayan.gopalappa/.nvm/versions/node/v6.0.0/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 545, in script_main
    return main(sys.argv[1:])
  File "/Users/narayan.gopalappa/.nvm/versions/node/v6.0.0/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 538, in main
    return gyp_main(args)
  File "/Users/narayan.gopalappa/.nvm/versions/node/v6.0.0/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/__init__.py", line 523, in gyp_main
    generator.GenerateOutput(flat_list, targets, data, params)
  File "/Users/narayan.gopalappa/.nvm/versions/node/v6.0.0/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/generator/make.py", line 2170, in Gene
rateOutput
    part_of_all=qualified_target in needed_targets)
  File "/Users/narayan.gopalappa/.nvm/versions/node/v6.0.0/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/generator/make.py", line 795, in Write
    self.Pchify))
  File "/Users/narayan.gopalappa/.nvm/versions/node/v6.0.0/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/generator/make.py", line 1190, in WriteSources
    cflags = self.xcode_settings.GetCflags(configname)
  File "/Users/narayan.gopalappa/.nvm/versions/node/v6.0.0/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/xcode_emulation.py", line 551, in GetCflags
    archs = self.GetActiveArchs(self.configname)
  File "/Users/narayan.gopalappa/.nvm/versions/node/v6.0.0/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/xcode_emulation.py", line 420, in GetActiveArchs
    xcode_archs_default = GetXcodeArchsDefault()
  File "/Users/narayan.gopalappa/.nvm/versions/node/v6.0.0/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/xcode_emulation.py", line 118, in GetXcodeArchsDefault
    xcode_version, _ = XcodeVersion()
  File "/Users/narayan.gopalappa/.nvm/versions/node/v6.0.0/lib/node_modules/npm/node_modules/node-gyp/gyp/pylib/gyp/xcode_emulation.py", line 1265, in XcodeVersion
    version = re.match(r'(\d\.\d\.?\d*)', version).groups()[0]
AttributeError: 'NoneType' object has no attribute 'groups'
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/Users/narayan.gopalappa/.nvm/versions/node/v6.0.0/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:305:16)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:204:12)
gyp ERR! System Darwin 18.2.0
gyp ERR! command "/Users/narayan.gopalappa/.nvm/versions/node/v6.0.0/bin/node" "/Users/narayan.gopalappa/.nvm/versions/node/v6.0.0/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/narayan.gopalappa/Desktop/Me/Git/TestBLE/node_modules/xpc-connection
gyp ERR! node -v v6.0.0
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok 
npm WARN install:xpc-connection@0.1.4 xpc-connection@0.1.4 install: `node-gyp rebuild`
npm WARN install:xpc-connection@0.1.4 Exit status 1
Test@1.0.0 /Users/narayan.gopalappa/Desktop/Me/Git/TestBLE
└─┬ noble@1.9.1 
  ├── bplist-parser@0.0.6 
  └─┬ debug@2.2.0 
    └── ms@0.7.1 

npm WARN optional Skipping failed optional dependency /noble/bluetooth-hci-socket:
npm WARN notsup Not compatible with your operating system or architecture: bluetooth-hci-socket@0.5.1
npm WARN Test@1.0.0 No repository field.
npm WARN Test@1.0.0 No license field.

kevin-smets, 2019-04-09:
What is your node version? I had to downgrade to v8 for the install to work. However, it's useless after that since #834

dsteinman, 2019-05-27:
This fork is working on Mojave:

https://github.com/abandonware/noble

LukasBombach, 2019-06-05:
Confirming, this package fails on my Mojave machine, the other repo (which is on npm as `@abandonware/noble`) works

playground, 2019-08-29:
I was able to install with https://github.com/abandonware/noble, but it's not able to discover any peripheral while it works with https://github.com/thegecko/webbluetooth.

dsteinman, 2019-08-29:
@playground The older Arduino BLE libraries based on https://github.com/sandeepmistry/arduino-BLEPeripheral did not work for me either.  I ended up migrating to the Adafruit bluefruit library for all the bluetooth code, and that works good with abandonware/noble.  It even works on armv6 for raspberry pi zero support.

LukasBombach, 2019-08-29:
For what it's worth I am completely rewriting `noble` here https://github.com/LukasBombach/sblendid/ with a modern API and entirely in TypeScript and native code for the BLE connection. 

So far everything actually works already, but I am currently refactoring my code (so the current commit is probably bronen) and will release soon

LukasBombach, 2019-08-29:
@dsteinman @playground I'd be happy to give you a ping when I release and get your feedback on this

dsteinman, 2019-08-29:
@LukasBombach looks great, I've starred your repo.

Does it work on WIndows?  Cause there's a whole other noble project dedicated to that:

https://github.com/jasongin/noble-uwp

Dealing with this stuff has been just a damned nightmare.

LukasBombach, 2019-08-29:
@dsteinman yes it does. noble-uwp has actually been forked and improved and realease here:

https://github.com/Timeular/noble-winrt

And that is what I am using as basis for my windows support: 
https://github.com/LukasBombach/sblendid/tree/master/packages/adapter-node/native

LukasBombach, 2019-08-29:
The thing is, for both, mac and windows I am using native `noble` adapters and I have successfully tested both so far. For Linux I believe I have to fall back to the JS implementation unfortunately, but I'll have to see about that.

dsteinman, 2019-08-29:
That's terrific.  As long as you can do whatever noble is using (I think it's a USB compatibility layer) then I imagine it will be fine on Raspberry pi's.

LukasBombach, 2019-08-29:
Yeah! Imma see :)