noble/noble #876: Can't recieve notifications.

MeddahAbdellah, 2019-03-26:
Hello, I am trying to read notification data on Raspberry Pi 3, OS : Raspbian , I have the following code : 

    noble.on('discover', function(device) {
		if(device.advertisement.localName=="resncid"){
		  console.log("Device Found : "+ device.advertisement.localName);
		  noble.stopScanning();
		  device.connect(function(error){console.log("error : "+error);});
		  device.once('connect', function(){
			  console.log("Connected");
			  device.discoverSomeServicesAndCharacteristics(serviceUUIDs, characteristicUUIDs, function(error, services, characteristics){
				console.log("discovered");
				//console.log(characteristics[0]);
				characteristics[0].on('data', function(data, isNotification){
					console.log("recieved : "+data);
				});
				characteristics[0].notify(true, function(error) {
				console.log('temperature notification on');
				})
				//console.log(characteristics);
				});
		  });
		}
	});
  noble.startScanning(); // any service UUID, no duplicates

carterharrison, 2019-03-28:
Having the same issue. Any result @MeddahAbdellah ? Make sure you .subscribe to the characteristic.

MeddahAbdellah, 2019-03-28:
@carterharrison I have tryed .subscribe, but no luck so far. I forgot to mention that I am using with node version 8.15.0 if that's of any importance.

DreammasterK, 2019-04-03:
I have the same problem as you. If you solve it,could you tell me how to do it.

dsteinman, 2019-05-27:
I had the same problems, and was able to locate a potential problem with the HCI gatt library, some detail here:

https://github.com/abandonware/noble/issues/15

MeddahAbdellah, 2019-05-30:
I haven't been able to fix it :/

dsteinman, 2019-05-31:
I was able to get subscribe and notify working by abandoning the firmware library I was using.

This library should be deprecated, it hasn't received any updates in over a year, and from the looks of it is a dead project that is now incompatible with noble:

https://github.com/sandeepmistry/arduino-BLEPeripheral

I'm switching to the NRF52 for BLE communication and the Bluefruit libraries do seem to be working with noble:

https://learn.adafruit.com/bluefruit-nrf52-feather-learning-guide/introduction