noble/noble #877: Is there a way to get the MTU of a connection?

carterharrison, 2019-03-28:
I am using noble to send kilobytes of data between phones so chunking of the data is required. Currently, I can only send 20 bytes at a time because I do not know how to get the MTU from a device using noble. I understand that callback for a MTU changes in the bindings [here](https://github.com/noble/noble/blob/master/lib/hci-socket/bindings.js#L273). Is there a reason why this is not exposed in the peripheral object? If not, I am happy to make the change and submit a PR. Thanks!

Mike-Dax, 2019-04-08:
Piggybacking on this issue, is there any method of knowing when we have written 'too much' to the device? NodeJS streams have this notion of a highWaterMark, when we're filling the buffer faster than the OS can deplete it. Is there any analog that's possible with BLE?

EricDobyns, 2019-06-17:
Added a PR to add the retrieved MTU value to the peripheral object. Also temporarily forked the repository here: https://github.com/EricDobyns/noble