noble/noble #880: Reading the Mi Bluetooth Temperature & Humidity Sensor

muscaiu, 2019-05-08:
Hello.

I'm currently using this python method (https://zsiti.eu/xiaomi-mijia-hygrothermo-v2-sensor-data-on-raspberry-pi/) to read the sensor. It works by giving it the mac address, but after a while it crashes and **i hate it cose it's python**. It's the only working way i could find to read the sensor.

What i'm looking for is, a noble or nodejs tutorial on how to **read the** [Mi Bluetooth Temperature & Humidity Sensor](https://www.xiaomistore.pk/mi-bluetooth-temperature-humidity-monitor.html) **in a nodejs app.**

Thanks.

[image]

micovi, 2020-10-26:
@muscaiu did you manage to read data from this with javascript? I am able to read the data from .advertisment on 'fe95' range but I can't make it subscribe or read data from any characteristic, they don't seem to change at all.

ChrisScheffler, 2021-01-11:
@muscaiu @micovi You can try my library [mipacket](https://github.com/ChrisScheffler/mipacket). These sensors transmit their values for temperature, humidity and battery via BLE advertisements. You don't need to connect to the sensors.

muscaiu, 2021-02-09:
2 years passed and i forgot how to scan for bluetooth mac adress.
I just bought the mi sensor v2 and been looking for a library to do it for 2 hours with no luck.
Any ideas?

rzr, 2021-02-15:
It looks the maintenance of this project is in pause, but feel free to contribute to @abandonware fork until upstream is active again

Feel free to comment:
abandonware#2