noble/noble #887: characteristic.subscribe() hitting error

aloebs29, 2019-05-30:
I'm on windows 10, using the asus BT-400.

I've successfully connected to my device, read all characteristics/properties, and written to a characteristic. However, I have a characteristic that I'd like to subscribe to for notifications (I have verified that this characteristic has the notify property), and I'm running into an error.

When I follow the code execution, it looks like within gatt.js at the notify function, I'm getting an op code associated with an error.

```
Gatt.prototype.notify = function(serviceUuid, characteristicUuid, notify) {
  var characteristic = this._characteristics[serviceUuid][characteristicUuid];

  this._queueCommand(this.readByTypeRequest(characteristic.startHandle, characteristic.endHandle, GATT_CLIENT_CHARAC_CFG_UUID), function(data) {
    var opcode = data[0];

    if (opcode === ATT_OP_READ_BY_TYPE_RESP) {
      var type = data[1];
      var handle = data.readUInt16LE(2);
      var value = data.readUInt16LE(4);

      var useNotify = characteristic.properties & 0x10;
      var useIndicate = characteristic.properties & 0x20;

      if (notify) {
        if (useNotify) {
          value |= 0x0001;
        } else if (useIndicate) {
          value |= 0x0002;
        }
      } else {
        if (useNotify) {
          value &= 0xfffe;
        } else if (useIndicate) {
          value &= 0xfffd;
        }
      }

      var valueBuffer = new Buffer(2);
      valueBuffer.writeUInt16LE(value, 0);

      this._queueCommand(this.writeRequest(handle, valueBuffer, false), function(data) {
        var opcode = data[0];

        if (opcode === ATT_OP_WRITE_RESP) {
          this.emit('notify', this._address, serviceUuid, characteristicUuid, notify);
        }
      }.bind(this));
    }
  }.bind(this));
};
```

The value of "data" in the callback is: { 1, 8, 54, 0, 10 }.

Any help would be much appreciated!

dsteinman, 2019-05-31:
Are you by chance using the [BLEPeripheral](https://github.com/sandeepmistry/arduino-BLEPeripheral) library ?

aloebs29, 2019-05-31:
No, I'm talking to a Microchip RN4678. I have confirmed with nRF connect on my android device that I can indeed subcribe to this notification and receive data.

ninggeng, 2019-09-18:
I have the same problem as you. If you solve it,could you tell me how to do it.

aloebs29, 2019-09-18:
I didn't find a fix, I ended up switching to web-bluetooth for my application. Here is the github thread where I asked the same question (in case it helps) -- 

https://stackoverflow.com/questions/56386067/noble-ble-module-characteristic-subscribe-hitting-error