noble/noble #895: Can't discover peripheral on OSX

ervinewell, 2019-08-01:
I create a BLE central with lightBlue app, and I can discover this on another android phone, but cannot discover this with noble as following:

`noble = require('noble')

noble.on('discover', function(peripheral) {
  console.log(1234)
});

noble.on('stateChange', function(state) {
console.log(state)
  if (state === 'poweredOn') {
    noble.startScanning([], true);
  } else {
    noble.stopScanning();
  }
});`

OSX: 10.14
node: 10.16.0

jaewonha, 2019-08-02:
I used https://github.com/Timeular/noble-mac as an alternative and its working

ervinewell, 2019-08-06:
> I used https://github.com/Timeular/noble-mac as an alternative and its working

It works. Thanks a lot.

closed 2019-08-06 by ervinewell