noble/noble #896: BLE supporting on Windows

ervinewell, 2019-08-05:
My Windows 7 PC supports bluetooth 4.0:
[image]

But when I use noble, it throws an error:
[image]

Are USB adaptors always necessary whatever bluetooth 4.0 is supported in PC? 
If so, is there anyway to use bluetooth 4.0 without a USB adaptor on a ble-native-supprted device?

n3rd4i, 2019-08-14:
You need to do 2 things:
From [here](https://github.com/noble/noble#windows):
* [WinUSB](https://msdn.microsoft.com/en-ca/library/windows/hardware/ff540196(v=vs.85).aspx) driver setup for Bluetooth 4.0 USB adapter, using [Zadig tool](http://zadig.akeo.ie/)
* Be sure to make a back-up of the original driver before replacing it, provided you will use this [WinUSB] driver instead.

Edit the allowed dongles, from [here](https://github.com/noble/node-bluetooth-hci-socket#compatible-bluetooth-40-usb-adapters):
* https://github.com/noble/node-bluetooth-hci-socket/blob/master/lib/usb.js#L14-L27

thefakhreddin, 2022-01-07:
@sandeepmistry 
This package works just fine on OS X with no problem. But the driver twitching on windows with Zadig is a real bummer. This makes the package impractical on windows for production purposes since it completely messes with the client's Bluetooth driver. (to the extent that the Bluetooth option completly vanishes from the taskbar and no one would want that.) 
Is there any solution or workaround not to change the adapter's driver on windows? Just like the way it works on Mac. In the future updates maybe?
Thanks.