noble/noble #897: BluetoothHciSocket Error?

nardev, 2019-08-09:
I don't think i'm the only one with this issue but i just can't find the solution.  I tried install on several different linux computers and few RPIes. 

Any solution?

`../src/BluetoothHciSocket.cpp:275:72: error: no matching function for call to ‘v8::Function::NewInstance(int, v8::Local<v8::Value> [1])’
   Local<Value> error = errorConstructor->NewInstance(1, constructorArgs);`

`/usr/include/nodejs/deps/v8/include/v8.h:9502:16: warning: cast between incompatible function types from ‘v8::WeakCallbackInfo<Nan::ObjectWrap>::Callback’ {aka ‘void (*)(const v8::WeakCallbackInfo<Nan::ObjectWrap>&)’} to ‘Callback’ {aka ‘void (*)(const v8::WeakCallbackInfo<void>&)’} [-Wcast-function-type]
make: *** [binding.target.mk:101: Release/obj.target/binding/src/BluetoothHciSocket.o] Error 1
make: Leaving directory '/home/pi/bletest/node_modules/bluetooth-hci-socket/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:262:23)
gyp ERR! stack     at ChildProcess.emit (events.js:189:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:248:12)
gyp ERR! System Linux 4.19.50+
gyp ERR! command "/usr/bin/node" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/bletest/node_modules/bluetooth-hci-socket
gyp ERR! node -v v10.15.2
gyp ERR! node-gyp -v v3.8.0
gyp ERR! not ok 
npm notice created a lockfile as package-lock.json. You should commit this file.`

omayhemo, 2019-08-13:
I've seen this issue widely reported since node 10.  Though it seems some forks have been made in an effort to correct the no longer maintained dependency causing the problem, none have solved the problem for me on 12.8

nardev, 2019-08-13:
Exactly, looks like the https://github.com/abandonware/noble fork is live and worth using.

omayhemo, 2019-08-13:
As you can you see by my [issue submission](https://github.com/abandonware/noble/issues/21) at abandonware, it doesn't work on 12.8

nardev, 2019-08-14:
> As you can you see by my [issue submission](https://github.com/abandonware/noble/issues/21) at abandonware, it doesn't work on 12.8

Well, at least some things are being maintained there. The minimum of the project lives there. It's not that big deal to maintain older version with nvm.