noble/noble #901: Impossible to install noble

Ravaniss, 2019-09-09:
```
npm i noble

> usb@1.6.0 install /home/max/Devs/test/node_modules/usb
> prebuild-install --verbose || node-gyp rebuild

prebuild-install info begin Prebuild-install version 5.3.1
prebuild-install info looking for cached prebuild @ /home/max/.npm/_prebuilds/2e98b1-usb-v1.6.0-node-v72-linux-x64.tar.gz
prebuild-install info found cached prebuild 
prebuild-install info unpacking @ /home/max/.npm/_prebuilds/2e98b1-usb-v1.6.0-node-v72-linux-x64.tar.gz
prebuild-install info unpack resolved to /home/max/Devs/test/node_modules/usb/build/Release/usb_bindings.node
prebuild-install info unpack required /home/max/Devs/test/node_modules/usb/build/Release/usb_bindings.node successfully
prebuild-install info install Successfully installed prebuilt binary!

> bluetooth-hci-socket@0.5.1 install /home/max/Devs/test/node_modules/bluetooth-hci-socket
> node-gyp rebuild

make : on entre dans le répertoire « /home/max/Devs/test/node_modules/bluetooth-hci-socket/build »
  CXX(target) Release/obj.target/binding/src/BluetoothHciSocket.o
../src/BluetoothHciSocket.cpp: In static member function ‘static void BluetoothHciSocket::Init(Nan::ADDON_REGISTER_FUNCTION_ARGS_TYPE)’:
../src/BluetoothHciSocket.cpp:128:82: error: no matching function for call to ‘v8::FunctionTemplate::GetFunction()’
 get->Set(Nan::New("BluetoothHciSocket").ToLocalChecked(), tmpl->GetFunction());
                                                                             ^
In file included from /home/max/.cache/node-gyp/12.10.0/include/node/node.h:63:0,
                 from /home/max/.cache/node-gyp/12.10.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/home/max/.cache/node-gyp/12.10.0/include/node/v8.h:5961:46: note: candidate: v8::MaybeLocal<v8::Function> v8::FunctionTemplate::GetFunction(v8::Local<v8::Context>)
   V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
                                              ^~~~~~~~~~~
/home/max/.cache/node-gyp/12.10.0/include/node/v8.h:5961:46: note:   candidate expects 1 argument, 0 provided
../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::poll()’:
../src/BluetoothHciSocket.cpp:251:95: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated [-Wdeprecated-declarations]
 back(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
                                                                              ^
In file included from ../src/BluetoothHciSocket.cpp:8:0:
../../nan/nan.h:1045:46: note: declared here
   NAN_DEPRECATED inline v8::Local<v8::Value> MakeCallback(
                                              ^~~~~~~~~~~~
../src/BluetoothHciSocket.cpp: In member function ‘void BluetoothHciSocket::emitErrnoError()’:
../src/BluetoothHciSocket.cpp:269:109: warning: ‘v8::Local<v8::Value> v8::Object::Get(v8::Local<v8::Value>)’ is deprecated: Use maybe version [-Wdeprecated-declarations]
 or = Local<Function>::Cast(globalObj->Get(Nan::New("Error").ToLocalChecked()));
                                                                             ^
In file included from /home/max/.cache/node-gyp/12.10.0/include/node/v8-internal.h:14:0,
                 from /home/max/.cache/node-gyp/12.10.0/include/node/v8.h:25,
                 from /home/max/.cache/node-gyp/12.10.0/include/node/node.h:63,
                 from /home/max/.cache/node-gyp/12.10.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/home/max/.cache/node-gyp/12.10.0/include/node/v8.h:3425:51: note: declared here
   V8_DEPRECATED("Use maybe version", Local<Value> Get(Local<Value> key));
                                                   ^
/home/max/.cache/node-gyp/12.10.0/include/node/v8config.h:311:3: note: in definition of macro ‘V8_DEPRECATED’
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
../src/BluetoothHciSocket.cpp:275:72: error: no matching function for call to ‘v8::Function::NewInstance(int, v8::Local<v8::Value> [1])’
   Local<Value> error = errorConstructor->NewInstance(1, constructorArgs);
                                                                        ^
In file included from /home/max/.cache/node-gyp/12.10.0/include/node/node.h:63:0,
                 from /home/max/.cache/node-gyp/12.10.0/include/node/node_buffer.h:25,
                 from ../src/BluetoothHciSocket.cpp:7:
/home/max/.cache/node-gyp/12.10.0/include/node/v8.h:4144:44: note: candidate: v8::MaybeLocal<v8::Object> v8::Function::NewInstance(v8::Local<v8::Context>, int, v8::Local<v8::Value>*) const
   V8_WARN_UNUSED_RESULT MaybeLocal<Object> NewInstance(
                                            ^~~~~~~~~~~
/home/max/.cache/node-gyp/12.10.0/include/node/v8.h:4144:44: note:   candidate expects 3 arguments, 2 provided
/home/max/.cache/node-gyp/12.10.0/include/node/v8.h:4147:44: note: candidate: v8::MaybeLocal<v8::Object> v8::Function::NewInstance(v8::Local<v8::Context>) const
   V8_WARN_UNUSED_RESULT MaybeLocal<Object> NewInstance(
                                            ^~~~~~~~~~~
/home/max/.cache/node-gyp/12.10.0/include/node/v8.h:4147:44: note:   candidate expects 1 argument, 2 provided
../src/BluetoothHciSocket.cpp:282:93: warning: ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated [-Wdeprecated-declarations]
 back(Nan::New<Object>(this->This), Nan::New("emit").ToLocalChecked(), 2, argv);
                                                                              ^
[... 63 lines cut ...]
    ^

Error: Cannot find module 'graceful-fs'
Require stack:
- /usr/share/node-gyp/lib/node-gyp.js
- /usr/share/node-gyp/bin/node-gyp.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:772:15)
    at Function.Module._load (internal/modules/cjs/loader.js:677:27)
    at Module.require (internal/modules/cjs/loader.js:830:19)
    at require (internal/modules/cjs/helpers.js:68:18)
    at Object.<anonymous> (/usr/share/node-gyp/lib/node-gyp.js:12:10)
    at Module._compile (internal/modules/cjs/loader.js:936:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:947:10)
    at Module.load (internal/modules/cjs/loader.js:790:32)
    at Function.Module._load (internal/modules/cjs/loader.js:703:12)
    at Module.require (internal/modules/cjs/loader.js:830:19) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/usr/share/node-gyp/lib/node-gyp.js',
    '/usr/share/node-gyp/bin/node-gyp.js'
  ]
}
```

I tried to remove the `package-lock.json`, `node_modules`, `node_gyp` but same problem...

janhaa, 2019-09-16:
Duplicate of https://github.com/noble/node-bluetooth-hci-socket/issues/84

Solution:

Ravaniss, 2019-09-19:
Thanks for your answer, i found @abandonware/noble and bleno, so it's perfect now. ^^

closed 2019-09-19 by Ravaniss

Liu-Jinshuai, 2023-09-12:
```
npm i bleno2
```