noble/noble #908: Enable caching of GATT characteristics

svogl, 2019-10-11:
We have a number of known BLE devices the we (re-)connect to often. Speeding up the connection setup phase is good for the user experience (less delay, less potential for packet loss), so we have extended noble a little to report the GATT data structures for services and characteristics and to initialize its inner workings with this data.

In order to not change default behavour, there are two new events one can listen for when discovering GATT entries, simply register them before the actual discovery call:
``` 
 peripheral.on('servicesDiscovered', (peripheral, services) => { ... }
 service.on('characteristicsDiscovered', (characteristics) => {...}
```
returns an array of services resp. characteristics including low-level information that can be used to initialize noble: 
```
 const servs = [
        { "startHandle": 1, "endHandle": 4, "uuid": "1801" },
        { "startHandle": 5, "endHandle": 9, "uuid": "1800" },
        { "startHandle": 10, "endHandle": 29, "uuid": "180a" }]
 let serviceObjs =  noble.addServices(peripheralUuid, servs)

 const enviroChara = [
  { "startHandle": 53, "properties": 18, "valueHandle": 54, "uuid": "2a6d", "propsDecoded": ["read", "notify"], "rawProps": 18, "endHandle": 57 },
  { "startHandle": 58, "properties": 18, "valueHandle": 59, "uuid": "2a6e", "propsDecoded": ["read", "notify"], "rawProps": 18, "endHandle": 62 }]

 let characteristicsObjs = noble.addCharacteristics(peripheral.uuid, enviroService.uuid, enviroChara);
```
 
addServices returns an array of service objects, addCharacteristics an array of Characteristic objects that  can be used to discover, subscribe for data etc.

The implementation is a small patch on top of the @abandonware/noble fork and can be found in this branch: https://github.com/svogl/noble/tree/inject-gatt-descriptors
and in this pull request: https://github.com/abandonware/noble/pull/29

Feedback & experiences highly welcome

pursual, 2020-09-09:
Doesnt work with the mac bindings. 
The mac bindings do not call `characteristicsDiscovered`. They also do not have `this._bindings.addCharacteristics` or `this._bindings.addServices`

svogl, 2020-09-15:
@pursual correct, mac support has not been implemented yet, we tried to improve performance for our project only; feel free to send patches!

pursual, 2020-09-15:
Would love this feature in the winrt bindings, but not sure it's possible to create a service object in windows, without actually discovering it. https://github.com/Timeular/noble-winrt/issues/20