noble/noble #913: send large data on bluetooth

Puneetkumarpal, 2019-10-29:
i want to send large data on Bluetooth 1 MB - 2 MB of data and take 3 - 4 min on SPP profile.
using service:- 0xFFE0 and profile;-0xFFE1 

i am using windows 10 and create application on electron using noble.

when i try to send large data it disconnected automatically please help me as soon as posible

Puneetkumarpal, 2019-11-01:
@sandeepmistry can u help me on this

joelviel, 2020-06-18:
I am currently testing noble, it can receive more than 20 kbytes in less than 2 seconds from Android phone. Noble should subscribe to peripheral notifications on a specific characteristic and recompose the data after receiving each notification (notification size limited to MTU size)