noble/noble #917: Multiple Disconnect Events

diginfo, 2019-11-27:
Device is RPI4 / O/S is Debian.

I am reading senors once a minute.

Each time I make a connection and then disconnect after reading data, the disconnected event fires once for each connection made to the same device, even though the previous connection was disconnected.

So the first time the connect is called, the disconnect event fires once, after calling the connect & disconnect 10 times one minute apart, I will get 10 disconnect events one after the other.

Any idea as to the cause ??

cyberfacture, 2020-01-22:
Hello,
I am having the same issue. Have you found a solution?

jjakob666, 2020-08-16:
Same thing is hapenning for me. and I know what is causing it but don't know how to fix it.  When you connect to a device, the profile is retained for the next connection even though you disconnect and reconnect again. So when you register dev.on('disconnect' every time you reconnect every 1 minute, you are actually adding the event listener again to the profile every time you call it.  thus have multiple events registered.  What I don't know how to do is to unregister the 'disconnect' event. if you can do that after receiving the event, then you won't have duplicate events fire. Same thing goes for noble.on('discover') this actually causes a memory leak if you register this event everytime you stop and start listening.

jjakob666, 2020-08-16:
I think the resolution is dev.on('disconnect',onDisconnect);
function onDisconnect() {
   //cleanup listener after disconnect
   dev.off('disconnect',onDisconnect);
}

hothsolo, 2022-06-09:
I'm having a similar problem here.  I did implement the fix above and that stopped my peripheral from repeating the connection over and over.  But occasionally, something will go haywire during the service or characteristic connection in my code.  If I restart the noble scanning, I'll get double console.log I'm throwing when I connect to the service and the characteristic.  And at that point nothing works.  Not sure why it's crashing on those calls, or how to do reset on them to get them to work again once they crash.  Any help is appreciated.

Cj