noble/noble #918: Does not install correctly - Cannot find module 'xpc-connection' or 'bluetooth-hci-socket'

mrkking, 2019-12-03:
### Steps to reproduce
npm install noble
create index.js file with contents:
    `const Noble = require('noble'); `
    `Noble.startScanning([], true);`

node index.js

 Xcode already installed
 OS: MAC Catalina
 Node: tried version 13.2.0, 13.1.0, 10.11.0, 9.xx.xx (last in version)
 tried to install xpc-connection - no luck

Also
 Raspberry Pi 3:
 tried same node versions as above
 tried to install bluetooth-hci-socket - no luck

### Error:
`Error: Cannot find module 'xpc-connection' - on MAC`
`Error: Cannot find module 'bluetooth-hci-socket'` - on RPI3

none of these libraries build properly xpc-connection or bluetooth-hci-socket for this project

ierome, 2019-12-16:
Refer to theGecko forked version of noble. Worked for me.

jakehawkes, 2020-01-14:
Just chiming in: on OSX the error is:

  CXX(target) Release/obj.target/binding/src/XpcConnection.o
../src/XpcConnection.cpp:26:77: error: too few arguments to function call, single argument 'context' was not
      specified
  target->Set(Nan::New("XpcConnection").ToLocalChecked(), tmpl->GetFunction());
                                                          ~~~~~~~~~~~~~~~~~ ^
/Users/jake/Library/Caches/node-gyp/13.6.0/include/node/v8.h:6283:3: note: 'GetFunction' declared here
  V8_WARN_UNUSED_RESULT MaybeLocal<Function> GetFunction(
  ^

jakehawkes, 2020-01-14:
> theGecko

And how would one find that?

rzr, 2020-01-14:
Any patch from @theGecko to be picked ?

thegecko, 2020-01-15:
The best version of noble today seems to be:

https://github.com/abandonware/noble

closed 2024-08-06 by mrkking