noble/noble #919: Connection Problem: "noble: unknown peripheral" on Raspberry Pi

Anonymussus, 2019-12-11:
I run Noble on the Raspberry Pi 3 (with integrated Bluetooth) and i am trying to get the measurements from a BLE scale.
The connection succeds only a very few times, mostly i get the "noble: unkown peripheral fb95560a1c5c" error message. I tried noble.StopScanning(); after my device is found like some people suggested in other threads, but i didn't help.

This is the used code:
`
var noble = require('./index');

noble.on('stateChange', function(state) {
  if (state === 'poweredOn') {
	noble.startScanning();
  } else {
    noble.stopScanning();
  }
});

noble.on('discover', function(peripheral) {
  if(peripheral.uuid = 'fb95560a1c5c'){
	  noble.stopScanning();
  }
  
  peripheral.connect(function(error) {
    console.log('Connected to peripheral: ' + peripheral.uuid + ', Status: ' + peripheral.state);

    peripheral.discoverServices(['181b'], function(error, services) {

      var BCSService = services[0];
      console.log('Discovered Service: ' + BCSService);

      BCSService.discoverCharacteristics(['2a9b'], function(error, characteristics) {

        var BCSCharacteristic = characteristics[0];
        console.log('Discovered Characteristic: ' + BCSCharacteristic);

        BCSCharacteristic.read(function(error, data) {
		console.log(data.toString('utf8'));								
		}

        //Notification
        BCSCharacteristic.subscribe(function(error) {
          console.log('Notification on');
        });
      });
    });
  });
});
`

YoRab, 2019-12-17:
Hello Anonymussus

Here you missed a double equal with you condition : 

> noble.on('discover', function(peripheral) {
> if(peripheral.uuid = 'fb95560a1c5c'){
> noble.stopScanning();
> }

Also, you try next line to connect the peripheral without checking the uuid first. Is it normal ?

rzr, 2020-01-18:
Try:
https://github.com/abandonware/noble