noble/noble #923: Use https://github.com/abandonware/noble instead!

eugef, 2020-01-19:
If you have issue with latest versions of Node.js or MacOs, you can consider using https://github.com/abandonware/noble instead!
It is a fork of this project which is actively maintained.

This package is not maintained for more than 4 years and doesn't have all the latest fixes.

rzr, 2020-02-19:
Would upstream be interested to merge changes from @abandonware org ? 

More about us:

https://abandonware.github.io/

rzr, 2020-02-20:
I just released a version:
https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-6

And will forward patches upstream:
https://github.com/abandonware/noble/issues/2

hobbs-sj, 2020-03-22:
Hi,
Any plans for Android support?

rzr, 2020-03-23:
You can open a ticket on:
https://github.com/abandonware/noble/

And see if there is any traction, I am wondering if tmux is also providing similar interfaces as linux..

rzr, 2020-04-03:
Is anyone in touch with author ?

May community contact upstream:

https://opencollective.com/noble/contact

rzr, 2020-04-03:
New version released thanks @rsmeral

https://github.com/abandonware/noble/pull/67

Some interesting stats:

https://libraries.io/npm/@abandonware%2Fnoble/usage

https://libraries.io/npm/noble/usage

More about us please help to promote about why we're doing this:

https://twitter.com/RzrFreeFr/status/1243569839886696451

rzr, 2020-05-14:
Please test:
https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-9

Please review:
https://github.com/noble/noble/pull/926

hobbs-sj, 2020-05-17:
Hi
I will do this but it will be a week or so before I can do it.

Many thanks.

Regards

Steve Hobbs
+44 7412880580

https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-9

#926<https://github.com/noble/noble/pull/926>

rzr, 2020-09-06:
https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-10 has been released

Please check if upstream's reported issues are fixed in this fork

rzr, 2020-09-18:
abandonware's branch is 104 commits ahead of noble:master. 

Please merge this first batch of changes:

https://github.com/noble/noble/pull/926

Then I'll rebase and send a 2d batch of changes.

rzr, 2021-03-30:
Please also check with:

https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-13

Relate-to: https://github.com/noble/noble/pull/926