noble/noble #924: Not able to connect BLE application with passcode enabled

sujitpatil72, 2020-01-28:
I am trying to connect Bluetooth device from my electron application where the passcode is enabled and am not getting below details

Can nobel take care of showing the dialogue to enter passcode? if YES, is there any callback where I can see get to know if the user entered valid passcode or wrong passcode?

If the passcode is not supported is there any way workaround?

projectbtle, 2021-02-28:
Can you try the suggestion in https://github.com/noble/noble/issues/117#issuecomment-787484119 ?
It will display a prompt to enter passkey. You can modify the code to perform additional checks prior to proceeding.

sujitpatil72, 2021-03-04:
Great!!, I will check, Thank you!!