noble/noble #929: Raspberry Pi 4 Support?

javiermelendezc, 2020-03-23:
Is the noble library compatible with the Raspberry 4? I know that RPi 4 comes with a BLE 5.0 module. In theory BLE 5.0 supports BLE 4.0, but is this library ready for this kind of support? Appreciate your feedback!

rzr, 2020-04-03:
Please try this version:
https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-8

And report issues also to:
https://github.com/abandonware/noble/issues

javiermelendezc, 2020-04-03:
Hey @rzr
Ended up getting a RPi 3 B+, and switching to @abandonware/noble
I reported an issue regarding its support on 
https://github.com/abandonware/noble/issues

I’m gonna close this one, to concentrate related issues and discussion in @abandonware/noble

closed 2020-04-03 by javiermelendezc