noble/noble #930: How noble is compatible with the new macOS(10.15 Catalina)?

lwj06, 2020-03-31:


rzr, 2020-04-03:
Please try this version:
https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-8

And report issues also to:
https://github.com/abandonware/noble/issues

lwj06, 2020-04-07:
> Please try this version:
> https://www.npmjs.com/package/@abandonware/noble/v/1.9.2-8
> 
> And report issues also to:
> https://github.com/abandonware/noble/issues

I didn't use this version before. Every time when the Mac OS system upgrade, I just need to change the kCBMsgId corresponding to the API in noble to be compatible with the new version of the system. In this 10.15.4, I found that the kCBMsgId was also changed, but the main problem is that the discoverservices callback didn't bring the servicesuuid parameter

closed 2020-11-04 by lwj06