noble/noble #931: noble connection Bluetooth fails at macOs catalina 10.15.4

lwj06, 2020-04-01:
[image]

every time Apple upgrades the system,it may cause noble can not compatible with the new macOs system. we can modify the msgId in /noble/lib/mac/xx.js 

I have found the correct msgId including :
stateChange: 6
startScanning:  53
response to startScanning: 60
scanStop:52
connect:  56
response to connect:    63
response to disconnect:  61
discoverServices:    102
response to discover service: 107

but I found out the parameter 'args' in 
nobleBindings.on('kCBMsgId107', function(args) {
...
})
is different from the previous version system,this different leads to an error in 
this.emit('servicesDiscover', deviceUuid, serviceUuids)

lwj06, 2020-04-01:
本来按照遍历通信id的方法，可以找出本次升级的系统，noble改变了的通信id有哪些，并且已经找出一部分msgId了，但是发现找到Response to discover service 这块功能的时候，找到的msgId=107是正确的，但是它接收到 discoverServices方法传过来的参数却跟以前的macOs版本不一样：
以前args是这样的：
{
  "kCBMsgArgServices": [{
    "kCBMsgArgServiceStartHandle": 11,
    "kCBMsgArgUUID": {
      "type": "Buffer",
      "data": [66, 48, 0, 1, 0, 35, 75, 212, 187, 213, 166, 146, 14, 76, 86, 83]
    },
    "kCBMsgArgServiceEndHandle": 65535
  }],
  "kCBMsgArgDeviceUUID": {
    "type": "Buffer",
    "data": [50, 102, 178, 225, 204, 136, 67, 235, 135, 253, 3, 164, 175, 172, 84, 132]
  },
  "kCBMsgArgResult": 0
}
现在的macOs 10.15.4版本是这样的：
{
	"kCBMsgArgDeviceUUID": {
		"type": "Buffer",
		"data": [100, 193, 154, 214, 15, 145, 125, 156, 214, 35, 54, 73, 154, 232, 56,134]
	},
	"kCBMsgArgResult": 312
}
一些参数缺失导致后面的流程报错了，后面的msgId都没办法继续找下去了，整个noble模块的兼容
最新版本的macOs系统也没办法做了

有人遇过这个问题吗？怎么解决的

dtknowlove, 2020-09-28:
@lwj06 解决了吗？同问，什么时候能有mac15的catalina.js?

lwj06, 2020-11-04:
> @lwj06 解决了吗？同问，什么时候能有mac15的catalina.js?

我直接换了包，用了：@abandonware/noble   1.9.2-8

closed 2020-11-04 by lwj06