noble/noble #935: different Node.js version

daniel-lo, 2020-04-28:
Hi, is there anything I can do about this?

(928) Noble not available, Error: Error: The module '/opt/iobroker/node_modules/@abandonware/bluetooth-hci-socket/lib/binding/binding.node'was compiled against a different Node.js version usingNODE_MO

node -v
v12.16.2
nodejs -v
v12.16.2
npm -v
6.14.4

ggiese2000, 2020-05-06:
+1 - any ETA if noble works again with the latest node versions?

rzr, 2020-05-13:
Please test latest abandonware's release and report if bug is also present in fork:

https://libraries.io/npm/@abandonware%2Fnoble/usage

daniel-lo, 2020-05-14:
I had tried that .. 16 days ago .. if I see correctly that covers the latest release.

RcTomcat1, 2020-06-09:
Any update on the matter? I encountered the same problem and am currently looking for a fix.

trogper, 2021-08-15:
@RcTomcat1 as rzr said, use [@abandonware/noble](https://www.npmjs.com/package/@abandonware/noble) instead of plain `noble`. I have tested it now on Rasberry Pi 2, Debian 10 and Node 14