noble/noble #937: Incorrect reading frames from Raspberry

mateok, 2020-05-05:
I have a problem with the library. After installing the latest version of Raspbian on Raspberry (I did tests about a year / two years ago and did not notice this problem), from time to time the library receives an incorrect frame. I have altbeacon devices that broadcast frames. Every once in a while (e.g. once every half hour, hour) I get the wrong frame and even the wrong RSSI value (although this is not always the case, though it usually occurs together). I use event: discover and receive data:
peripheral.advertisement.manufacturerData.toString ('hex')
and rssi:
peripheral.rssi
Sometimes rssi is positive (e.g. 30, and it should be around -70, -60). There is a wrong frame with rssi, e.g. my device gives a fixed value:
5900beacdba5190470d94f9294f004c02145cc29c908182ac3f4
And once in a while he receives:
5900beacdba5190470d94f9294f004c02145cc29c908c300f4a3
Where do I know that the device did not give a 100% other value. What could the problem be?

mateok, 2020-05-05:
I have more logs, this time the whole peripheral object (here you can see that RSSI is wrong, the value in plus and data in manufacturerData is less:
`{"id":"c0730d2f3666","address":"c0:73:0d:2f:36:66","addressType":"random","connectable":false,"advertisement":{"manufacturerData":{"type":"Buffer","data":[89,0,190,172,219,165,25,4,112,217,79,146,148,240,4,192,33,69,204,35,224,178,203,62,195,192]},"serviceData":[],"serviceUuids":[],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]},"rssi":-60,"state":"disconnected"}`

Here is the correct reading:
`{"id":"c0730d2f3666","address":"c0:73:0d:2f:36:66","addressType":"random","connectable":false,"advertisement":{"manufacturerData":{"type":"Buffer","data":[89,0,190,172,219,165,25,4,112,217,79,146,148,240,4,192,33,69,204,35,224,178,203,62,4,0]},"serviceData":[],"serviceUuids":[],"solicitationServiceUuids":[],"serviceSolicitationUuids":[]},"rssi":62,"state":"disconnected"}`

Of course, I could verify the correctness of RSSI, but I know that it will do about 90% of cases, but sometimes RSSI is ok and the reading is incorrect.

rzr, 2020-05-07:
to be compared with:

https://github.com/noble/noble/issues/923

airdrummingfool, 2021-05-08:
@mateok did you ever figure this out? What Pi did you see this on?

I'm experiencing the same issue on a Raspberry Pi 3. I have not yet seen this problem on my RPi 4s, so I'm wondering if there's a race condition somewhere (or the BLE hardware in the Pi 3 is struggling).

Making sure the RSSI is negative seems to be a good sanity check - every time it is positive, the scan response seems to be corrupted. I also added a check for my expected manufacturer ID (the first 2 bytes of the scan response), which catches a lot of corruption.