noble/noble #938: Cannot install noble in RPI 3b

danelo11, 2020-05-05:
Here is the outpot of doing npm install noble --save:

Debugger listening on ws://127.0.0.1:44357/b68c850d-8015-49b1-8526-599c313e35ec
For help, see: https://nodejs.org/en/docs/inspector
Debugger attached.

> usb@1.6.3 install /home/pi/pfg/node_modules/usb
> prebuild-install --verbose || node-gyp rebuild

Debugger listening on ws://127.0.0.1:35825/e189759b-a734-480d-8304-c1c639ee8f11
For help, see: https://nodejs.org/en/docs/inspector
Debugger attached.
prebuild-install info begin Prebuild-install version 5.3.3
prebuild-install info looking for cached prebuild @ /home/pi/.npm/_prebuilds/470134-usb-v1.6.3-node-v72-linux-arm.tar.gz
prebuild-install http request GET https://github.com/tessel/node-usb/releases/download/v1.6.3/usb-v1.6.3-node-v72-linux-arm.tar.gz
prebuild-install http 404 https://github.com/tessel/node-usb/releases/download/v1.6.3/usb-v1.6.3-node-v72-linux-arm.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=12.16.3 runtime=node arch=arm libc= platform=linux)
Waiting for the debugger to disconnect...
Debugger listening on ws://127.0.0.1:42879/3e2b41e1-3a71-4146-8f20-f44d69ceab7f
For help, see: https://nodejs.org/en/docs/inspector
Debugger attached.
Debugger listening on ws://127.0.0.1:41121/faaf0657-ba67-482f-9a92-703d3e2a06c6
For help, see: https://nodejs.org/en/docs/inspector
Debugger attached.
Waiting for the debugger to disconnect...
gyp: Call to 'node -e "require('nan')"' returned exit status 0 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:351:16)
gyp ERR! stack     at ChildProcess.emit (events.js:310:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Linux 4.19.97-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/pfg/node_modules/usb
gyp ERR! node -v v12.16.3
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
Waiting for the debugger to disconnect...

> bluetooth-hci-socket@0.5.1 install /home/pi/pfg/node_modules/bluetooth-hci-socket
> node-gyp rebuild

Debugger listening on ws://127.0.0.1:34047/e9bdd6e7-e68b-41c2-9711-e4038072fbe1
For help, see: https://nodejs.org/en/docs/inspector
Debugger attached.
Debugger listening on ws://127.0.0.1:36909/d2f2b0a8-58ea-4fbc-8db9-95dfb9524c54
For help, see: https://nodejs.org/en/docs/inspector
Debugger attached.
Waiting for the debugger to disconnect...
gyp: Call to 'node -e "require('nan')"' returned exit status 0 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/configure.js:351:16)
gyp ERR! stack     at ChildProcess.emit (events.js:310:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Linux 4.19.97-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/pfg/node_modules/bluetooth-hci-socket
gyp ERR! node -v v12.16.3
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok 
Waiting for the debugger to disconnect...
Debugger listening on ws://127.0.0.1:34047/e9bdd6e7-e68b-41c2-9711-e4038072fbe1
For help, see: https://nodejs.org/en/docs/inspector
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: xpc-connection@~0.1.4 (node_modules/noble/node_modules/xpc-connection):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for xpc-connection@0.1.4: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN pfg@1.0.0 No description
npm WARN pfg@1.0.0 No repository field.
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.6.3 (node_modules/usb):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: usb@1.6.3 install: `prebuild-install --verbose || node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 (node_modules/bluetooth-hci-socket):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: bluetooth-hci-socket@0.5.1 install: `node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

+ noble@1.9.1
added 4 packages from 4 contributors and audited 110 packages in 34.036s

1 package is looking for funding
  run `npm fund` for details

found 2 low severity vulnerabilities
  run `npm audit fix` to fix them, or `npm audit` for details
Waiting for the debugger to disconnect...

----------------------------------------------
Thank you in advance, I'm really looking forward to your reply.

rzr, 2020-05-07:
may this help:

https://github.com/noble/noble/issues/923

danelo11, 2020-05-07:
It works for me!
Thank you very much
Shall I close this issue?

rzr, 2020-05-07:
you can keep it open and request upstream to merge patches at:

https://github.com/noble/noble/pulls

Review welcome there too

rzr, 2020-09-06:
Relate-to: https://github.com/noble/noble/issues/923#issuecomment-687851954