noble/noble #948: Xiaomi temperature and humidity sensor 2

ChrisWoehrle, 2020-07-24:
Is it possible to control xiaomi temperature and humidity sensor 2? There is no active blutooth connection.

oyooyo, 2020-08-30:
Not sure what exactly you mean by "there is no active bluetooth connection".

Which one is the Xiaomi temperature and humidity sensor 2, the round one or the square one? The round one sends the temperature and humidity data as part of the advertising packets, so one does not even have to establish a bluetooth connection.

The square one on the other hand sends the temperature and humidity data via characteristic _ebe0ccbe-7a0a-4b0c-8a1a-6ff2997da3a6_ (service _ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6_).
humidity = [third byte] %
temperature = (([second byte] * 256) + [first_byte]) / 100)°C
Strangely, even though this characteristic supports both read and subscribe, I only get valid data when using subscribe. When using read instead, I only get 0s.

noelhibbard, 2021-01-20:
The default firmware for this device encrypts that data. There is 3rd party firmware that fixes this issue. Check this out: https://github.com/atc1441/ATC_MiThermometer

rzr, 2021-01-26:
It looks the maintenance of this project is in pause, but feel free to contribute to @abandonware fork until upstream is active again

Feel free to comment:
https://github.com/abandonware/noble/issues/2

muzea, 2021-05-31:
Of course, the current temperature data is on the broadcast packet. You can check [the implementation here](https://github.com/leonxi/mitemp_bt2/blob/main/custom_components/mitemp_bt2/sensor.py), see how to parse the data.