noble/noble #951: No compatible USB Bluetooth 4.0 device found

ummarbhutta, 2020-10-28:
I am evaluating this library for one of my project, to communicate with Bluetooth LE devices. Following are details of my system

`
Windows 10 64bit OS (Build 18363.1082)
HP ProBook 450 G7
USB VID 8087
USB PID 0AAA
`
I did all prerequisites for library to run on windows, but when I execute command `node advertisement-discovery.js` I get following error

`C:\Users\noble-master\examples>node advertisement-discovery.js
C:\Users\noble-master\node_modules\bluetooth-hci-socket\lib\usb.js:70
    throw new Error('No compatible USB Bluetooth 4.0 device found!');
    ^

Error: No compatible USB Bluetooth 4.0 device found!
    at BluetoothHciSocket.bindUser (C:\Users\noble-master\node_modules\bluetooth-hci-socket\lib\usb.js:70:11)
    at BluetoothHciSocket.bindRaw (C:\Users\noble-master\node_modules\bluetooth-hci-socket\lib\usb.js:28:8)
    at Hci.init (C:\Users\bakmal\Downloads\noble-master\lib\hci-socket\hci.js:101:35)
    at NobleBindings.init (C:\Users\noble-master\lib\hci-socket\bindings.js:82:13)
    at Noble.<anonymous> (C:\Users\noble-master\lib\noble.js:57:24)
    at processTicksAndRejections (internal/process/task_queues.js:79:11)`

any idea why it is so, Is it due to reason that it is not in compatible USB list [here](https://github.com/noble/node-bluetooth-hci-socket#windows)?

Secondly I have a question, Why there is dependency on `Zadig`, Is this library usable without it? Or we have to add this additional step in production environment (not a good approach I believe).?

partyzonetut, 2020-11-17:
Same device and problem

sirShakir, 2021-03-30:
Im running ubuntu and having the same issue. Any solutions?