noble/noble #953: BLE Adapter unauthorised in Docker Container

popeyesfx, 2020-12-01:
I am running node red in a docker container and when I start I get the following:

1 Dec 17:28:45 - [info] Starting flows
1 Dec 17:28:45 - [info] Started flows
noble warning: adapter state unauthorized, please run as root or with sudo
               or see README for information on running without root/sudo:
               https://github.com/sandeepmistry/noble#running-on-linux
1 Dec 17:28:45 - [info] [BLE scanner:1810f4b7.a85c7b] BLE state: unauthorized
1 Dec 17:28:46 - [info] [bmp180:BMP180 @ 0x77] BMP180 @ 0x77 ready.
1 Dec 17:28:46 - [info] Server now running at http://127.0.0.1:1880/
1 Dec 17:29:44 - [error] [BLE scanner:1810f4b7.a85c7b] Error scanning for devices: Error: Could not start scanning, state is unauthorized (not poweredOn)

I have tried to follow the running as user advice on the page but it does not work inside the container.

The Docker run command looks like this:

docker run -it --privileged -v node_red_data:/data --device /dev/i2c-1 --net=host --restart="always" --group-add 998 --name nodered nodered/node-red:1.2.0-10-arm32v6

Any help would be appreciated

stich86, 2020-12-06:
I’ve solved forcing user “root” on the container instead of “node red” but I want to run the process without root account

toufou54, 2022-08-07:
Same issue for me without docker, even with the setcap command on node binary and node-pi binary (I have installed node-red with the raspberry bash script recommended)

rzr, 2024-02-06:
May you forward the issue to @abandonware noble until upstream gives feedback at:

https://github.com/noble/noble/issues/656