noble/noble #980: Cannot find module 'xpc-connection'

jhackett1, 2021-12-08:
I have run `npm i noble` on a fresh node.js project on Mac OSX, with Xcode installed.

When I run this script:

```
const noble = require("noble")

noble.startScanning()
```

I get an error:

```
Error: Cannot find module 'xpc-connection'
Require stack:
- /Users/joshuahackett/Desktop/bermondsey/node_modules/noble/lib/mac/highsierra.js
- /Users/joshuahackett/Desktop/bermondsey/node_modules/noble/lib/mac/bindings.js
- /Users/joshuahackett/Desktop/bermondsey/node_modules/noble/lib/resolve-bindings.js
- /Users/joshuahackett/Desktop/bermondsey/node_modules/noble/index.js
- /Users/joshuahackett/Desktop/bermondsey/index.js
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:933:15)
    at Function.Module._load (node:internal/modules/cjs/loader:778:27)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:94:18)
    at Object.<anonymous> (/Users/joshuahackett/Desktop/bermondsey/node_modules/noble/lib/mac/highsierra.js:7:21)
    at Module._compile (node:internal/modules/cjs/loader:1101:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1153:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19) {
  code: 'MODULE_NOT_FOUND',
```

I've tried explicitly doing `npm i xpc-connection` with no luck.

What's up?

zbot473, 2021-12-08:
i'm pretty sure this repo is dead

gabe0x02, 2022-01-07:
Looks like this fork is more active: https://github.com/abandonware/noble