noble/noble #983: Driver swapping with Zadig is a real bummer in windows

thefakhreddin, 2022-01-08:
Hello noble contribiutors.
I'm using this module on Mac os with no problem at all. And I was also able to make it work on windows as well. The only problem is that in order for this module to function on windows properly, you have to swap the **bluetooth driver** with **Zadig** which practically makes the blutooth adaptor useless. E.g. you cannot turn your adaptor on and off in windows or connect your bluetooth headset.
This constraint makes this great module impractical for commercial usage since the client must willingly mess his/her computer's bluetooth and this is just too much to ask.

I know the techniques required to roll back to the standard driver whenever you want but this isn't something that any user would want.

Is there any solution or a workaround to avoid messing with the driver? Like OSX/ Mac.

Thanks.

RamlahAziz, 2022-01-22:
Unfortunately not. Stuck on this as well. I'm using the more maintained [@abandonware/bleno](https://github.com/abandonware/bleno) library to try and get it to run on windows but the code doesn't detect a stateChange. 

I'm trying to make it work on windows. I've replaced a Bluetooth 2.0 USB driver with Zadig WinUSB and the "No compatible Bluetooth 4.0 USB Device found error" is gone. The library detects a stateChange and emits it but the listener doesn't pick it up for some reason. I've tried running all the library examples, the noble/bleno examples and the [node-bluetooth-hci-socket ](https://github.com/noble/node-bluetooth-hci-socket) examples but nothing. It runs with no error but just doesn't pick up the stateChange.

Can you let me know the versions of node and node-gyp you are using on your windows machine?

JJTech0130, 2022-03-11:
I see that https://github.com/hbldh/bleak somehow manages to use BLE on Windows, without requiring you to replace the driver... perhaps something similar could be implemented here?

thefakhreddin, 2022-03-13:
@RamlahAziz 
Hello. I wish I saw your message earlier. Try using `theGecko/noble`. Add this line in `dependencies` in your packages.json:
`"noble": "theGecko/noble"`

thefakhreddin, 2022-03-13:
@JJTech0130 Thanks for your suggestion.
If only we could notify the contributors.
@sandeepmistry