noble/noble #987: error

shahinesi, 2022-03-17:
```
shahineskandari@shahins-MacBook-Pro ~ % npm audit fix --force
npm WARN using --force Recommended protections disabled.
npm WARN audit Updating noble to 0.0.2,which is a SemVer major change.
npm ERR! code 1
npm ERR! path /Users/shahineskandari/node_modules/noble
npm ERR! command failed
npm ERR! command sh -c node-gyp rebuild
npm ERR! CXX(target) Release/obj.target/binding/src/Noble.o
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@8.4.1
npm ERR! gyp info using node@16.14.1 | darwin | arm64
npm ERR! gyp info find Python using Python version 3.9.10 found at "/Library/Frameworks/Python.framework/Versions/3.9/bin/python3"
npm ERR! gyp info spawn /Library/Frameworks/Python.framework/Versions/3.9/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/shahineskandari/node_modules/noble/build/config.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/usr/local/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/Users/shahineskandari/Library/Caches/node-gyp/16.14.1/include/node/common.gypi',
npm ERR! gyp info spawn args   '-Dlibrary=shared_library',
npm ERR! gyp info spawn args   '-Dvisibility=default',
npm ERR! gyp info spawn args   '-Dnode_root_dir=/Users/shahineskandari/Library/Caches/node-gyp/16.14.1',
npm ERR! gyp info spawn args   '-Dnode_gyp_dir=/usr/local/lib/node_modules/npm/node_modules/node-gyp',
npm ERR! gyp info spawn args   '-Dnode_lib_file=/Users/shahineskandari/Library/Caches/node-gyp/16.14.1/<(target_arch)/node.lib',
npm ERR! gyp info spawn args   '-Dmodule_root_dir=/Users/shahineskandari/node_modules/noble',
npm ERR! gyp info spawn args   '-Dnode_engine=v8',
npm ERR! gyp info spawn args   '--depth=.',
npm ERR! gyp info spawn args   '--no-parallel',
npm ERR! gyp info spawn args   '--generator-output',
npm ERR! gyp info spawn args   'build',
npm ERR! gyp info spawn args   '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! In file included from ../src/Noble.cpp:4:
npm ERR! In file included from ../src/BLEManager.h:7:
npm ERR! ../src/Noble.h:11:21: error: expected class name
npm ERR! class Noble : node::ObjectWrap {
npm ERR!                     ^
npm ERR! ../src/Noble.h:23:24: error: no template named 'Handle' in namespace 'v8'
npm ERR!   static void Init(v8::Handle<v8::Object> target);
npm ERR!                    ~~~~^
npm ERR! ../src/Noble.h:24:14: error: no template named 'Handle' in namespace 'v8'
npm ERR!   static v8::Handle<v8::Value> New(const v8::Arguments& args);
npm ERR!          ~~~~^
npm ERR! ../src/Noble.h:24:46: error: no type named 'Arguments' in namespace 'v8'
npm ERR!   static v8::Handle<v8::Value> New(const v8::Arguments& args);
npm ERR!                                          ~~~~^
npm ERR! ../src/Noble.h:26:14: error: no template named 'Handle' in namespace 'v8'
npm ERR!   static v8::Handle<v8::Value> StartScanning(const v8::Arguments& args);
npm ERR!          ~~~~^
npm ERR! ../src/Noble.h:26:56: error: no type named 'Arguments' in namespace 'v8'
npm ERR!   static v8::Handle<v8::Value> StartScanning(const v8::Arguments& args);
npm ERR!                                                    ~~~~^
npm ERR! ../src/Noble.h:27:14: error: no template named 'Handle' in namespace 'v8'
npm ERR!   static v8::Handle<v8::Value> StopScanning(const v8::Arguments& args);
npm ERR!          ~~~~^
npm ERR! ../src/Noble.h:27:55: error: no type named 'Arguments' in namespace 'v8'
npm ERR!   static v8::Handle<v8::Value> StopScanning(const v8::Arguments& args);
npm ERR!                                                   ~~~~^
npm ERR! ../src/Noble.h:29:14: error: no template named 'Handle' in namespace 'v8'
npm ERR!   static v8::Handle<v8::Value> ConnectPeripheral(const v8::Arguments& args);
npm ERR!          ~~~~^
npm ERR! ../src/Noble.h:29:60: error: no type named 'Arguments' in namespace 'v8'
npm ERR!   static v8::Handle<v8::Value> ConnectPeripheral(const v8::Arguments& args);
npm ERR!                                                        ~~~~^
npm ERR! ../src/Noble.h:30:14: error: no template named 'Handle' in namespace 'v8'
npm ERR!   static v8::Handle<v8::Value> DisconnectPeripheral(const v8::Arguments& args);
npm ERR!          ~~~~^
npm ERR! ../src/Noble.h:30:63: error: no type named 'Arguments' in namespace 'v8'
npm ERR!   static v8::Handle<v8::Value> DisconnectPeripheral(const v8::Arguments& args);
npm ERR!                                                           ~~~~^
npm ERR! ../src/Noble.h:32:27: error: unknown type name 'uv_work_t'
npm ERR!   static void UpdateState(uv_work_t* req);
npm ERR!                           ^
npm ERR! ../src/Noble.h:33:36: error: unknown type name 'uv_work_t'
npm ERR!   static void PeripheralDiscovered(uv_work_t* req);
npm ERR!                                    ^
npm ERR! ../src/Noble.h:34:35: error: unknown type name 'uv_work_t'
npm ERR!   static void PeripheralConnected(uv_work_t* req);
npm ERR!                                   ^
npm ERR! ../src/Noble.h:35:40: error: unknown type name 'uv_work_t'
npm ERR!   static void PeripheralConnectFailure(uv_work_t* req);
npm ERR!                                        ^
npm ERR! ../src/Noble.h:36:37: error: unknown type name 'uv_work_t'
npm ERR!   static void PeripheralDisonnected(uv_work_t* req);
npm ERR!                                     ^
npm ERR! In file included from ../src/Noble.cpp:4:
npm ERR! ../src/BLEManager.h:9:35: error: no type or protocol named 'CBCentralManagerDelegate'
npm ERR! @interface BLEManager : NSObject <CBCentralManagerDelegate> {
npm ERR!                                   ^
npm ERR! ../src/BLEManager.h:24:31: error: unknown type name 'CBCentralManager'
npm ERR! @property (nonatomic, retain) CBCentralManager *centralManager;
npm ERR!                               ^
npm ERR! fatal error: too many errors emitted, stopping now [-ferror-limit=]
npm ERR! 20 errors generated.
npm ERR! make: *** [Release/obj.target/binding/src/Noble.o] Error 1
npm ERR! gyp ERR! build error 
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:526:28)
npm ERR! gyp ERR! stack     at Process.ChildProcess._handle.onexit (node:internal/child_process:291:12)
npm ERR! gyp ERR! System Darwin 21.3.0
npm ERR! gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /Users/shahineskandari/node_modules/noble
npm ERR! gyp ERR! node -v v16.14.1
npm ERR! gyp ERR! node-gyp -v v8.4.1
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/shahineskandari/.npm/_logs/2022-03-17T06_43_43_070Z-debug-0.log
```

shahinesi, 2022-03-17:
```
shahineskandari@shahins-MacBook-Pro ~ % npm install noble              

up to date, audited 9 packages in 2s

2 low severity vulnerabilities

To address all issues (including breaking changes), run:
  npm audit fix --force

Run `npm audit` for details.
```

rzr, 2022-11-29:
Please check if problem is also present with downstream community maintained fork:

https://github.com/abandonware/noble

If yes please report issue there too